SiliconLabsSoftware/arduino #1: Wrong Commander_linux_x86_64_1v14p5b1276 for Linux

rei-vilo, 2024-01-09:
The https://siliconlabs.github.io/arduino/package_arduinosilabs_index.json mentions `Commander_linux_x86_64_1v14p5b1276` with the wrong extension, `zip`.
```
                {
                    "version": "1.14.5",
                    "name": "simplicitycommander",
                    "systems": [
                        {
                            "host": "x86_64-pc-linux-gnu",
                            "url": "https://github.com/SiliconLabs/arduino/releases/download/1.0.0/Commander_linux_x86_64_1v14p5b1276.zip",
                            "archiveFileName": "Commander_linux_x86_64_1v14p5b1276.zip",
                            "checksum": "SHA-256:9c81f401ff38d0eb8719389ea570ba3880852d099b6c8b9495c375cc442e0e9b",
                            "size": "58726426"
                        }
                    ]
                }
```

The correct address is https://github.com/SiliconLabs/arduino/releases/download/1.0.0/Commander_linux_x86_64_1v14p5b1276.tar.bz with `tar.bz`.

silabs-bozont, 2024-01-09:
Hello @rei-vilo!
Thanks for bringing this to my attention! You get the award for the first reported issue! 🏆
Indeed it was an issue on our end - I've updated the JSON file with the correct details, please give it a go again and let me know if it works!

rei-vilo, 2024-01-10:
You're welcome!

Installation completed successfully.

Case closed, thank you.

closed 2024-01-10 by rei-vilo