SiliconLabsSoftware/arduino #3: Button inverted logic in BLE_Blinky example

rei-vilo, 2024-01-13:
Shoudln't this line...

https://github.com/SiliconLabs/arduino/blob/956de5de292f53d907c2746cacfb5a26de3e49fb/libraries/SiliconLabs/examples/ble_blinky/ble_blinky.ino#L166

...be...

```
    btn_state = 1 - digitalRead(BTN_BUILTIN);
```

...to display a solid blue button on the smartphone app when pressed and an empty button otherwise?  

Thank you!

silabs-bozont, 2024-01-13:
Hello rei-vilo,
We have different boards with different wiring on the buttons. Some kits have a pullup resistor, others have a pulldown, so the unpressed state is different on them. The BGM220 Explorer Kit you have has a pullup resistor, so it's in the HIGH state when not pressed and LOW when pressed.
The example just means to show how to communicate a state change through BLE - we consider it to be working as long as the button press changes the state regardless of the starting state. But of course feel free to invert it for yourself if you like it better that way. The example is there to give you a starting point.
Of course we could've added an #ifdef for each board type - but the goal was to keep it simple.

rei-vilo, 2024-01-14:
Thank you for the confirmation.

closed 2024-01-14 by rei-vilo

silabs-bozont, 2024-02-02:
Hello rei-vilo,
Just an FYI - I took your advice and the button state will be inverted in the next release. Seems like I was mistaken and all boards have an active low logic, so it makes sense to invert. Thanks for shedding light on this.

rei-vilo, 2024-02-02:
Great!