SiliconLabsSoftware/arduino #4: Issue with debugging

rei-vilo, 2024-01-13:
I am using the _blinky_ example on the BGM220 Explorer kit. Build and link, upload and run work fine.

When trying to debug according to the [procedure](https://github.com/SiliconLabs/arduino#debugging), 

```
[2024-01-13T17:34:00.769Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
JLinkGDBServer -singlerun -nogui -if SWD -port 50000 -swoport 50001 -telnetport 50002 -device BGM220PC22HNA
[2024-01-13T17:34:05.970Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

The `debug_custom.json` file contains

```
{
  "servertype": "jlink",
  "device": "BGM220PC22HNA",
  "interface": "SWD",
  "serverpath": "JLinkGDBServer"
}
```

The installed JLinkGDBServer version is
 
> SEGGER J-Link GDB Server V7.94d Command Line Version
> JLinkARM.dll V7.94d (DLL compiled Jan 11 2024 09:42:08)

on 

> Linux NUC-12 6.1.0-17-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.69-1 (2023-12-30) x86_64 GNU/Linux

However, Simplicity Studio debugs fine.

What am I missing? Thank you!

silabs-bozont, 2024-01-19:
Hello rei-vilo,
Excuse me for my late reply, I needed some time to take a deeper dive on this.
I can confirm your issue - debugging is broken on Linux for me as well. The good news is that I found the issue and there's a workaround available.

When I tried to launch a debugging session I got the same error that 
`GDB server session ended. This terminal will be reused, waiting for next session to start...`
I tried launching `JLinkGDBServer` independently of the Arduino IDE - that worked fine, it recognized the connected MCU.
Then I tried connecting to it with a gdb session from the command line when I noticed that `arm-none-eabi-gdb` is crashing hard. This is what causes the Arduino IDE session to end prematurely.

I suggest you try and launch the supplied `arm-none-eabi-gdb` on your system and see if it crashes for you as well.
It's located at `/home/<user>/.arduino15/packages/SiliconLabs/tools/gcc-arm-none-eabi/12.2.rel1/bin/arm-none-eabi-gdb`.
When I launched it first it complained about missing an ncurses shared library. I solved this by installing the `ncurses5-compat-libs` package.
On the second launch I got further, now I had a python error:
```Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = '/usr/local/bld-tools/bld-tools-virtual-env/bin/python'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/usr/local/bld-tools/bld-tools-virtual-env/bin/python'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.executable = '/usr/local/bld-tools/bld-tools-virtual-env/bin/python'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/usr/lib/python38.zip',
    '/usr/lib/python3.8',
    '/usr/lib/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x000073ca44203c00 (most recent call first):
<no Python frame>
```

This seems to be a known issue with GDB hard requiring python3.8 to be in a specific location on Linux:
https://www.linuxquestions.org/questions/slackware-14/can%92t-run-arm%92s-gdb-python-problem-4175724434/
https://stackoverflow.com/questions/72187185/arm-none-eabi-gdb-cannot-start-because-of-some-python-error

I managed to work this around by installing python3.8.18 with pyenv and linking it to the location specified by the error message:
`pyenv install 3.8.18`
`sudo mkdir -p /usr/local/bld-tools/bld-tools-virtual-env/bin`
`sudo ln -s /home/<user>/.pyenv/versions/3.8.18/bin/python python`
After this `arm-none-eabi-gdb` launched without any errors and debugging could be successfully started from the Arduino IDE.

Let me know if this works out for you! 
If not - please post the output from running the built-in `arm-none-eabi-gdb` at `/home/<user>/.arduino15/packages/SiliconLabs/tools/gcc-arm-none-eabi/12.2.rel1/bin/arm-none-eabi-gdb` and we can dive further!
I know this is not very user friendly having to hack your system like this in order to use debugging - we'll try to figure something out for the next release to make this easier on Linux users.

rei-vilo, 2024-01-20:
Thank you for the thorough investigation and detailed instructions.

I installed the missing library with `sudo apt install libncursesw5`.

`JLinkGDBServer` starts fine when called outside the Arduino IDE. 

Calling it from the Arduino IDE still fails. It may be the same issue as reported at https://github.com/arduino/arduino-ide/issues/2115. Using the Arduino IDE FlatPak distribution limits access to external tools.  

Instead of `arm-none-eabi-gdb`, I am using `gdb-multiarch`. It allows to get rid of the Python limitation.

```
$ gdb-multiarch BG24_Blink_Debug.ino.elf
(gdb) target extended :50000
(gdb) load
(gdb) list loop
(gdb) br 34
(gdb) c
Continuing.

Thread 2 hit Breakpoint 1, loop ()
    at BG24_Blink_Debug.ino:34
34	  delay(100);                      // wait for a second
(gdb)
```

I managed to debug against the BGM220 and the BG24 Explorer boards using [emCode](https://rei-vilo.github.io/emCode/), a tool I have designed to run on Visual Studio Code.

For the BG24 Explorer board, please note that the recommended MCU `EFR32MG24B210F1536IM48` is unknown: `EFR32MG24BXXXF1536` seems to be the correct one.

The last board to perform tests is the xG27 Dev Kit. Is there a way to get a sample?

silabs-bozont, 2024-01-25:
That's a quite impressive project you got going on! Thanks for adding support for our boards quickly.
I'm glad you could work the issue around.
The `EFR32MG24B210F1536IM48` and `EFR32MG24BXXXF1536` are both seem to be accepted as targets by JLinkGDBServer - I just checked it. I'm on version V7.94g.
The xG27 Dev Kit does not support Matter, it's BLE only. It can be purchased from our [website](https://www.silabs.com/development-tools/wireless/efr32xg27-development-kit), but it's a bit pricey at 75 USD. Unfortunately development kits are not included in our [sample program](https://www.silabs.com/products/buy-sample/FAQS).

closed 2024-06-10 by rei-vilo