SiliconLabsSoftware/arduino #7: Matter and bluetooth simultaneously

OABsoftware, 2024-02-04:
I bought a Sparkfun Thing Plus Matter. And I love it!
One thing I'd like to use it for is to make a device, that has a bluetooth interface, Matter compatible. In this scenario the Thing would expose the Matter interface to Homekit and simultaneously it would communicate with de device I mentioned over bluetooth (BLE).
I'm using the Arduino IDE to develop this "man in the middle" scenario.

But this doesn't seem to be feasible. It looks like the Thing either supports Matter over Thread or it supports Bluetooth.

Is that correct? And, if so, why is that?

silabs-bozont, 2024-02-07:
Hello OABsoftware,
Happy that you're enjoying your board! :) I took a look at the current implementation and it looks like that [BLEManagerImpl](https://github.com/SiliconLabs/arduino/blob/main/variants/thingplusmatter_matter/matter_2.1.1/src/platform/silabs/efr32/BLEManagerImpl.cpp) takes ownership of the Bluetooth event loop - this means that with the current implementation it's not possible to run a custom BLE app besides Matter.
If you're using Simplicity Studio then it's different, you can generate an additional BLE event handler hook for your custom apps and share the stack that way. It would involve creating a custom component which registers a BLE event handler.
I think this feature is something that could be possibly added to the Arduino Core - I'll add this to our list of things to do and see if we can come up with something in the future.

OABsoftware, 2024-02-09:
I’ve tried Simplicity Studio and looked at both Matter and Bluetooth examples. But this has a pretty steep llearning curve. A bit too steep for me…
It would be great if simultaneous use of Matter and Bluetooth could be made available through the Arduino IDE!