SiliconLabsSoftware/arduino #11: Generic Switch device example

lboue, 2024-02-05:
Hello,

Switches are not supported in Apple Home. Could you add Generic Switch device example? 

Regards

silabs-bozont, 2024-02-05:
Hello Iboue,
Unfortunately switches are currently not supported by the major ecosystems including Apple Home. That's why the example has been left out. Once Switch support becomes available we will definitely include an example for it. The Switch class is available to experiment with, so if you have a Matter hub that supports it you can use it at your own risk :)

In the meantime if you want to achieve similar functionality - you can control your lightbulb's on/off state with a physical switch and sync the state to the Matter network, but it won't be able to control other devices in the same group as a regular Matter Switch would. There will also be an example for this in the future.

lboue, 2024-02-07:
I have the Philips Hue bridge with is exposing [Hue Dimmer Switch](https://www.philips-hue.com/fr-fr/p/hue-hue-dimmer-switch--modele-le-plus-recent-/8719514274617) via Matter protocol, so I it possible. It works with Apple Home:
[image]

lboue, 2024-02-07:
Here's a sample implementation:
[Generic Switch](https://github.com/espressif/esp-matter/tree/main/examples/generic_switch#22-momentary-switch).
_This example creates a Generic Switch device using the ESP Matter data model_

lboue, 2024-02-07:
> I have the Philips Hue bridge with is exposing [Hue Dimmer Switch](https://www.philips-hue.com/fr-fr/p/hue-hue-dimmer-switch--modele-le-plus-recent-/8719514274617) via Matter protocol, so I it possible to expose Momentary switch.

Here is the device exposed to Matter fabric:
```json
{
   "7/29/0":[
      {
         "deviceType":17,
         "revision":1
      },
      {
         "deviceType":19,
         "revision":1
      }
   ],
   "7/29/1":[
      29,
      57,
      47,
      46
   ],
   "7/29/2":[
      
   ],
   "7/29/3":[
      8,
      9,
      10,
      11
   ],
   "7/29/65532":0,
   "7/29/65533":1,
   "7/29/65528":[
      
   ],
   "7/29/65529":[
      
   ],
   "7/29/65531":[
      0,
      1,
      2,
      3,
      65528,
      65529,
      65531,
      65532,
      65533
   ],
   "7/57/5":"Hue dimmer salon",
   "7/57/17":true,
   "7/57/2":4107,
   "7/57/1":"Signify Netherlands B.V.",
   "7/57/3":"RWL021",
   "7/57/10":"1.1.28573",
   "7/57/18":"115dc69142c249e1a217447262f98823",
   "7/57/14":"Hue dimmer switch",
   "7/57/65532":0,
   "7/57/65533":1,
   "7/57/65528":[
      
   ],
   "7/57/65529":[
      
   ],
   "7/57/65531":[
      5,
      17,
      2,
      1,
      3,
      10,
      18,
      14,
      65528,
      65529,
      65531,
      65532,
      65533
   ],
   "7/47/14":0,
   "7/47/1":1,
   "7/47/0":1,
   "7/47/2":"Primary Battery",
   "7/47/25":3,
   "7/47/15":false,
   "7/47/16":2,
   "7/47/19":"Unspecified",
   "7/47/65532":10,
   "7/47/65533":1,
   "7/47/65528":[
      
   ],
   "7/47/65529":[
      
   ],
   "7/47/65531":[
      14,
      1,
      0,
      2,
      25,
      15,
      16,
      19,
      65528,
      65529,
      65531,
      65532,
      65533
   ],
   "7/46/0":[
      7
   ],
   "7/46/65532":0,
   "7/46/65533":1,
   "7/46/65528":[
      
   ],
   "7/46/65529":[
      
   ],
   "7/46/65531":[
      0,
      65528,
      65529,
      65531,
      65532,
      65533
   ]
}
```

lboue, 2024-02-07:
> Hello Iboue, Unfortunately switches are currently not supported by the major ecosystems including Apple Home. That's why the example has been left out. Once Switch support becomes available we will [...]

The is 2 [generic-switch certified products](https://www.matteralpha.com/category/generic-switch):
1. [Tuo Smart Button](https://www.matteralpha.com/tuo-accessories-llc/tuo-smart-button-p3) by TUO Accessories
2. [Kasa Matter Smart Dimmer Switch](https://www.matteralpha.com/kasa/kasa-smart-dimmer-switch-p11)
 by Kasa

silabs-bozont, 2024-02-08:
Thanks for the information - Button support now seems to be more widely available! I'll look into this and if possible we'll make the Switch examples available in the next release.

lboue, 2024-02-10:
I also found this tutorial from SL : [Matter over OpenThread Multi-feature Button](https://github.com/SiliconLabs/matter_applications/tree/master/matter_thread_multi_feature_switch_mad)

## Overview ##

This tutorial will show you how to create and build a general Matter switch app that can trigger multiple types of events using a single button. This is based on the Silicon Labs Matter Github (SMG) repository's existing light-switch app. 

For the demonstration purpose, I will use 2 x [EFR32xG24 Wireless 2.4 GHz +10 dBm Radio Board (BRD4186C)](https://www.silabs.com/development-tools/wireless/xg24-rb4186c-efr32xg24-wireless-gecko-radio-board?tab=overview) and [EFR32 Wireless Starter Kit (WSTK) Mainboard (BRD4001A)](https://www.silabs.com/development-tools/wireless/efr32xg24-pro-kit-10-dbm?tab=overview), one for the switch and the other for the light. You can use any eligible listed in the hardware required section.

Events that this button can trigger are:
- Short press
- Short release
- Long press
- Long release
- Multi press

silabs-bozont, 2024-02-15:
Thanks for the linked resources! I have good news - Matter Switch support will be coming soon (along with an example) in the next release.
[image]

lboue, 2024-02-15:
> Thanks for the linked resources! I have good news - Matter Switch support will be coming soon (along with an example) in the next release.
Nice! Will the new version be released in February?

silabs-bozont, 2024-02-15:
The new version is planned to be released around the 21st of March during [Arduino Days](https://days.arduino.cc/about) to support the new Nano Matter board.

lboue, 2024-03-21:
> The new version is planned to be released around the 21st of March during [Arduino Days](https://days.arduino.cc/about) to support the new Nano Matter board.

Could you tell us when will the new version be published?

closed 2024-05-07 by silabs-bozont