SiliconLabsSoftware/arduino #12: [MGM240P] issue with matter_lightbulb_multiple_color.ino -> Not responding

lboue, 2024-02-05:
Hello 
I am facing an issue with Matter matter_lightbulb_multiple_color.ino example. After pairing, device is not responding.

## Steps to reproduce

1. Flash matter_lightbulb_multiple_color.ino to SparkFun Thing Plus MGM240P board
2. Pair it with Apple Home
3. Check  Matter device status in Apple Home ==> Device is not responding

lboue, 2024-02-05:
For context, this is my setup: 
- Thead BR
  - 2 * AppleTV 4K
  - 1 HomePod mini
- two Nordic nRF5340 boards that work without a problem with Matter over Thread
- 2 * [Eve Energy](https://www.amazon.fr/Eve-Energy-Built-Schedules-Bluetooth/dp/B0929GGTJC) smartplug

lboue, 2024-02-05:
Same issue with another one: matter_lightbulb.ino

```
18:54:28.649 -> Matter lightbulb
19:00:47.749 -> Matter device is not commissioned
19:00:47.749 -> Commission it to your Matter hub with the manual pairing code or QR code
19:00:47.749 -> Manual pairing code: 34970112332
19:00:47.749 -> QR code URL: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A6FCJ142C00KA0648G00
19:04:13.475 -> Waiting for network connection...
19:04:14.438 -> Device connected
```

silabs-bozont, 2024-02-07:
Hello Iboue,
Glad it eventually started working - but you're right, it should not take 6 minutes to get synced. 
Verbose logs from the Matter and Thread stack are available through RTT. You can read them with Segger's [JLinkRTTViewer](https://www.segger.com/products/debug-probes/j-link/tools/rtt-viewer/). 
Start the RTT viewer and select the MGM240PB32VNA MCU:
[image]
Hit the reset on your board, then connect the RTT viewer to it. Your device should work as normal, wait for it to get synced and control the lightbulb a few times if possible. Please post the RTT logs from your device and we'll see what can cause it.

lboue, 2024-02-07:
>   Hit the reset on your board, then connect the RTT viewer to it. Your device should work as normal, wait for it to get synced and control the lightbulb a few times if possible. Please post the RTT logs from your device and we'll see what can cause it.

**J-Link RTT**
```
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]Arduino Matter device starting
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]Init CHIP Stack
[00:00:00.000][silabs ]Initializing OpenThread stack
[00:00:00.000][info  ][DL] OpenThread started: OK
[00:00:00.000][info  ][DL] Setting OpenThread device type to ROUTER
[00:00:00.000][silabs ]Starting OpenThread task
[00:00:00.000][info  ][SVR] Subscription persistence not supported
[00:00:00.000][info  ][SVR] Server initializing...
[00:00:00.000][detail][FP] Initializing FabricTable from persistent storage
[00:00:00.000][info  ][TS] Last Known Good Time: 2024-02-07T19:01:23
[00:00:00.000][info  ][DMG] Acce
```

silabs-bozont, 2024-02-08:
Thanks for the logs and the screenshots!
Looks like your RTT log is cut super short - it should print a LOT of messages. Please make sure that you first reset your board, then connect it to the RTT viewer. Once the board is connected reset shouldn't be issued as it will not automatically connect back.

lboue, 2024-02-09:
> Thanks for the logs and the screenshots! Looks like your RTT log is cut super short - it should print a LOT of messages. Please make sure that you first reset your board, then connect it to the RTT [...]

I'm going to do it again, but it takes a long time to compile. About 15 min.

lboue, 2024-02-09:
Is there an IPv6 packet size limitation in your implementation that would explain these errors?

```
Line  361: [00:03:50.129][error ][SVR] Server initialization complete
Line  424: [00:03:51.929][error ][BLE] no endpoint for unsub recvd
Line  974: [00:04:21.632][error ][DL] _WriteThreadNetworkDiagnosticAttributeToTlv failed: Error CHIP:0x0000000B
Line  975: [00:04:21.632][error ][DMG] Error retrieving data from clusterId: 0x0000_0035, err = b
Line 1003: [00:04:21.983][error ][DL] _WriteThreadNetworkDiagnosticAttributeToTlv failed: Error CHIP:0x0000000B
Line 1004: [00:04:21.984][error ][DMG] Error retrieving data from clusterId: 0x0000_0035, err = b
```

Fail
```
Line  939: [00:04:20.991][detail][DMG] <RE:Run> first cluster event is too big so that it fails to fit in the packet!
Line  974: [00:04:21.632][error ][DL] _WriteThreadNetworkDiagnosticAttributeToTlv failed: Error CHIP:0x0000000B
Line  977: [00:04:21.633][detail][DMG] <RE:Run> first cluster event is too big so that it fails to fit in the [00:04:21.951][detail][IN] UDP Message Received packet nb : 73 SrcAddr : fd4f:21ef:e564:0:83d0:2d1d:5d3f:c9b9[58997] DestAddr : fd4f:21ef:e564:0:b684:b5e0:3915:7a0a[5540] Payload Length 42
Line 1003: [00:04:21.983][error ][DL] _WriteThreadNetworkDiagnosticAttributeToTlv failed: Error CHIP:0x0000000B
Line 1005: [00:04:21.984[00:04:21.984][detail][DMG] <RE:Run> first cluster event is too big so that it fails to fit in the packet!
Line 1050: [00:04:23.044][detail][DMG] <RE:Run> first cluster event is too big so that it fails to fit in the packet!
```

lboue, 2024-02-09:
To compare, I build a Matter the from [SparkFun Thing Plus Matter - MGM240P Hookup Guide](https://learn.sparkfun.com/tutorials/sparkfun-thing-plus-matter---mgm240p-hookup-guide/all).
It works fine with the same board.

[matter_SimplicityStudio_lightbulb_JRTT.log](https://github.com/SiliconLabs/arduino/files/14224464/matter_SimplicityStudio_lightbulb_JRTT.log)

lboue, 2024-02-12:
Did you find the issue root cause?

lboue, 2024-03-25:
Since the new release is delayed by a month, could you help me correct this issue?

silabs-bozont, 2024-03-27:
This issue will most probably improved on in the next release, but a total fix won't be possible on the current version. Discovery time not only depends on the device, but the Matter hub as well. You could try "stimulating" your device from the app by clicking on it a few times or closing/opening the app. This workaround usually pushes the hub to discover the device quicker - you can see something happening in the RTT logs - sometimes I use this too, but YMMV.
However for me the devices were always discovered right after pairing - the delay only occurred on subsequent reboots.

lboue, 2024-03-27:
So why does it work well for examples built with Simplicity Studio? 
I only have this delay issue when I build with this Arduino library?

silabs-bozont, 2024-03-27:
The currently published Arduino Matter lags behind the currently released Matter SDK which is available through Studio - that's why there's a big difference between them.
The new release will update the Arduino Matter to the latest Matter SDK.

lboue, 2024-04-24:
@silabs-bozont 

Version 1.0 was released in January and nothing has happened since. When SL will release the update the Arduino Matter?

I check Mouser and [Arduino Nano Matter](https://store.arduino.cc/pages/nano-matter) is expected 25-Jul-24.

closed 2024-05-09 by lboue