SiliconLabsSoftware/arduino #13: BridgedDeviceBasicInformation cluster

lboue, 2024-02-07:
Hi,

Can you add [BridgedDeviceBasicInformation ](https://github.com/project-chip/connectedhomeip/blob/master/docs/clusters.md)cluster to you examples?

## List of currently defined clusters

| Code (dec) | Code (hex) | Name                                                    |
| ---------- | ---------- | ------------------------------------------------------- |
|          3 |       0x03 | Identify                                                |
|          4 |       0x04 | Groups                                                  |
|         .. |       ..   | ...				                                        |
|         **57** |       **0x39** | **BridgedDeviceBasicInformation**                           |
|         59 |       0x3B | Switch                                                  |

This makes it possible to present a comprehensible device name in the user interface.

silabs-bozont, 2024-02-07:
Hello lboue,
Thank you, great suggestion! This is coming in the next version along with functions for customizing your devices, like setting it's name/manufacturer/serial.

lboue, 2024-02-07:
> Hello lboue, Thank you, great suggestion! This is coming in the next version along with functions for customizing your devices, like setting it's name/manufacturer/serial.

Great!

silabs-bozont, 2024-02-08:
Here's a little sneak peek:
[image]

lboue, 2024-02-08:
> You'll be able to set these values to anything you'd like.
> 
> 

It's perfect!

silabs-bozont, 2024-05-07:
It's now available in 2.0.0

closed 2024-05-07 by silabs-bozont