SiliconLabsSoftware/arduino #14: Debuging with Arduino IDE fails

lboue, 2024-02-09:
Debuging with Arduino IDE fails with this error:

```
Invalid servertype parameters. The following values are supported: "jlink", "openocd", "stlink", "stutil", "pyocd", "bmp", "pe", "qemu", "external"
```

**debug_custom.json**
{
    "servertype": "jlink",
    "device": "MGM240PB32VNA",
    "interface": "SWD",
    "serverpath": "C:/Program Files/SEGGER/JLink_V794d/JLinkGDBServerCL.exe"
}

silabs-bozont, 2024-02-10:
Hey lboue,
I've just checked - debugging on Windows works for me without any issues - on the Arduino IDE 2.2.1.
[image]

Looks like Arduino IDE 2.3.0 breaks debugging compatibility with the current methods - I get the same error as you did with the setup that's worked on 2.2.1. We'll have to look into what's changed and update the readme.

In the meantime if you'd like to debug I'd advise you to temporarily downgrade to 2.2.1. Once I'll know the changes and the method that works on 2.3.0 I'll let you know.

silabs-bozont, 2024-02-10:
I've found a workaround - you'll need to change a config value in the **launch.json** file:
Select the debugging icon form the lefthandside menu:
[image]

Click the cogwheel icon - then the **launch.json** file will be opened in the editor.
[image]

[image]
In the config named **Arduino** change the **servertype** parameter to **jlink** - and that's it. This gets rid of the error and lets me debug the code.

Let me whether this worked for you!

silabs-bozont, 2024-02-10:
Also - this will be fixed in the newer releases so users won't have to change this manually.

lboue, 2024-02-10:
> Let me whether this worked for you!

```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "configId": "SiliconLabs:silabs:thingplusmatter_matter:programmer=commander",
      "cwd": "${workspaceRoot}",
      "request": "launch",
      "type": "cortex-debug",
      "executable": "C:\\Users\\INTEL-N100\\AppData\\Local\\Temp\\arduino\\sketches\\3EAA7A8A625EA056DE8D67FA7B43AFE6/matter_lightbulb.ino.elf",
      "toolchainPrefix": "arm-none-eabi",
      "name": "SparkFun Thing Plus Matter (Matter) (commander)",
      "servertype": "jlink",
      "armToolchainPath": "C:\\Users\\INTEL-N100\\AppData\\Local\\Arduino15\\packages\\SiliconLabs\\hardware\\silabs\\1.0.0/../../../tools/gcc-arm-none-eabi/12.2.rel1/bin/",
      "configFiles": [
        ""
      ],
      "device": "MGM240PB32VNA",
      "interface": "swd"
    }
  ]
}
```

I am getting this:
```
Waiting for gdb server to start...[2024-02-10T14:38:48.453Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
JLinkGDBServerCL.exe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MGM240PB32VNA
[2024-02-10T14:38:53.795Z] SERVER CONSOLE DEBUG: GDBServerConsole: onBackendConnect: gdb-server program client error Error: read ECONNRESET
[2024-02-10T14:38:53.804Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

[image]

TomInIowa, 2024-02-14:
"in the config named Arduino change the servertype parameter to jlink. "

When I click on debug,   arduino puts the file back to what it was.
How to save the change?

lboue, 2024-02-16:
@TomInIowa Did you try with Arduino IDE 2.3.1?

silabs-bozont, 2024-02-16:
Until the upstream fix is published I'll help you folks to work around the issue in the current release. If you do these steps you'll be able to debug with the newer 2.3.x IDEs.

First close the Arduino IDE. 

Secondly you'll need to modify *platform.txt* in your core installation. It's located in the [Arduino15](https://support.arduino.cc/hc/en-us/articles/360018448279-Open-the-Arduino15-folder) folder which is at different places on different OSs.
**macOS:** /Users/<your_user>/Library/Arduino15/packages/SiliconLabs/hardware/silabs/1.0.0
**Linux:** /home/<your_user>/.Arduino15/packages/SiliconLabs/hardware/silabs/1.0.0
**Windows:** c:\Users\<your_user>\AppData\Local\Arduino15\packages\SiliconLabs\hardware\silabs\1.0.0
Open platform.txt and change the following lines:
```
debug.server=jlinkgdbserver
debug.server.jlinkgdbserver.path=
debug.server.jlinkgdbserver.scripts_dir=
debug.server.jlinkgdbserver.script=
```
to
```
debug.server=jlink
debug.server.jlink.path=
debug.server.jlink.scripts_dir=
debug.server.jlink.script=
```

You can open your IDE back up. It's time to modify the **debug_custom.json** file. The new IDE version requires a new format which looks like this:
```
[
    {
        "configId": "SiliconLabs:silabs:thingplusmatter_ble:programmer=commander",
        "servertype": "jlink",
        "device": "MGM240PB32VNA",
        "interface": "swd",
        "serverpath": "JLinkGDBServer"
    }
]
```
Change **serverpath** according to you platform and debugger install location.
Change **device** according to your MCU.

Change the `(variant name)` in the **configId** property according to the FQBN (Fully Qualified Board Name) of the variant you're using:
*"configId": SiliconLabs:silabs:`(variant name)`:programmer=commander*

Here's a table listing all the `variant name`s:

| Board                                         | Variant name                    |
| -------------                                 |:-------------:|
| SparkFun ThingPlus Matter (BLE)               | thingplusmatter_ble             |
| SparkFun ThingPlus Matter (BLE) (precomp)     | thingplusmatter_ble_precomp     |
| SparkFun ThingPlus Matter (Matter)            | thingplusmatter_matter          |
| SparkFun ThingPlus Matter (Matter) (precomp)  | thingplusmatter_matter_precomp  |
| xG27 Dev Kit (BLE)                            | xg27devkit                      |
| xG27 Dev Kit (BLE) (precomp)                  | xg27devkit_precomp              |
| xG24 Explorer Kit (BLE)                       | xg24explorerkit                 |
| xG24 Explorer Kit (BLE) (precomp)             | xg24explorerkit_precomp         |
| xG24 Explorer Kit (Matter)                    | xg24explorerkit_matter          |
| xG24 Explorer Kit (Matter) (precomp)          | xg24explorerkit_matter_precomp  |
| BGM220 Explorer Kit (BLE)                     | bgm220explorerkit               |
| BGM220 Explorer Kit (BLE) (precomp)           | bgm220explorerkit_precomp       |

Make sure you select **Simplicity Commander** as your programmer.
After these steps debugging should work for you in the Arduino IDE 2.3.x and onwards.

TomInIowa, 2024-02-16:
Thank you   silabs-bozont.     Debugger is working.

I am running Arduino IDE
Date: 2024-02-15T08:55:06.887Z (1 day ago)
CLI Version: 0.35.2

per1234, 2024-03-03:
There is a bug in Arduino IDE >=2.3.0 that causes the values of certain keys in `debug_custom.json` to not take effect:

https://github.com/arduino/vscode-arduino-tools/issues/48

The affected keys include the `[*].servertype` and `[*].serverpath` keys used by this platform.

You can work around the bug by using these alternative key names:

|Original key|Workaround key|
|-|-|
| `armToolchainPath` | `toolchainPath`|
| `configFiles` | `serverConfiguration.scripts`|
| `serverpath` | `serverPath`|
| `servertype` | `server`|

With the workaround keys added, the `debug_custom.json` file contents would look like this:

```json
[
    {
        "configId": "SiliconLabs:silabs:thingplusmatter_ble:programmer=commander",
        "server": "jlink",
        "servertype": "jlink",
        "device": "MGM240PB32VNA",
        "interface": "swd",
        "server": "JLinkGDBServer",
        "serverpath": "JLinkGDBServer"
    }
]
```

Even though the `[*].servertype` and `[*].serverpath` keys currently have no effect, I left them in the file so that it will continue to work even after the IDE bug is fixed.

silabs-bozont, 2024-03-07:
Hi @per1234! Thank you very much for the heads up and the workaround!

per1234, 2024-03-07:
You are welcome. Reviewing my previous comment, I see I made an error in the `debug_custom.json` snippet I shared. `"server": "JLinkGDBServer",` should have been `"serverPath": "JLinkGDBServer",`:

```json
[
    {
        "configId": "SiliconLabs:silabs:thingplusmatter_ble:programmer=commander",
        "server": "jlink",
        "servertype": "jlink",
        "device": "MGM240PB32VNA",
        "interface": "swd",
        "serverPath": "JLinkGDBServer",
        "serverpath": "JLinkGDBServer"
    }
]
```

TomInIowa, 2024-03-18:
are there other Silicon Labs boards that support Arduino Debugging just **like** the MGM240 board?   maybe the SparkFun Thing Plus - ESP32-C6???

Callum6677, 2024-03-19:
> are there other Silicon Labs boards that support Arduino Debugging just **like** the MGM240 board? maybe the SparkFun Thing Plus - ESP32-C6???

The esp32-c6 is espressif not silicon labs.

silabs-bozont, 2024-03-20:
Yes, the SparkFun Thing Plus ESP32-C6 while being the same form factor - is not from us, it's from Espressif. EFR32 and ESP32 are just one letter off :)
However all of the supported Silicon Labs boards have debugging capability - even the new Nano Matter will offer it too.

closed 2024-05-08 by lboue