SiliconLabsSoftware/arduino #15: Matter over Thread

TomInIowa, 2024-02-11:
I am on a Mac Computer.  Love the MGM240P board.   Going thru the different Matter examples.     I also have some EveHome devices with the Eve app on my ipad.  The Eve app scans for Thread devices.   It does NOT find the MGM240P board!   
Current Library is NOT matter over thread?????

Brezensalzer, 2024-02-12:
Hi Tominiowa,
I have two SparkFun MGM240P boards as a Matter sensor (temperature, humidity, pressure) and a Matter rgb light bulb. Both are commissioned in my Thread network (Homepod Mini as border router) and work flawlessly. I also have two EVE Energy plugs and an EVE Light Switch in my Matter /Thread fabric.
The EVE app shows both MGM240P boards as Router (Full Thread Devices) in my Thread network, but it needs some time (up to a minute) until they show up.

TomInIowa, 2024-02-13:
I used the same board ....   also used Arduino IDE and the Silicon Lab  matter library.   Started with the Example Matter Light Color.     Did you?
I tried the Simplicity Studio 5 but never figured out how to get the code commision it so I could  add it to google matter and/or Apple HomeKit

silabs-bozont, 2024-02-13:
Hello TomInIowa,
Happy to hear that you're enjoying your SparkFun board!
As @Brezensalzer said - it's a (Full) Thread Device, so you're getting Matter over Thread indeed.
A few things to check:
- Did you burn the bootloader after switching to the Matter variant in Arduino IDE? Matter variants use a different bootloader than BLE variants.
- Do you have any output on the Serial port?
- You can check for the device's BLE advertisement whether it's present when in pairing mode
- You can check the RTT logs with SEGGER's RTT viewer as described [here](https://github.com/SiliconLabs/arduino/issues/12#issuecomment-1932497990)

Let me know if checking these helped. If not - please provide your Serial/RTT outputs so we can look further into the issue.

TomInIowa, 2024-02-14:
I erased the board using Silicon Commander.
Uploaded the ino and then the bootloader

Maybe I am using the "wrong" library.    If I try to tell Arduino to use the zip library on https://github.com/SiliconLabs/arduino. I tells me it is not a valid library.        What I did is add https://siliconlabs.github.io/arduino/package_arduinosilabs_index.json to arduino ide, then tell it I have a SparkFun board, then arduino installed the Silicon Labs matter things.

Using the arduino ide output window, I get the maual pairing code and the url.  Then Apple homekit will add it as an accessory.

I am trying to learn about the RTT logs and how to view them.

TomInIowa, 2024-02-16:
debug problem solution is in           Debuging with Arduino IDE 2.3.0 fails #14

TomInIowa, 2024-02-17:
[Rttlog.txt](https://github.com/SiliconLabs/arduino/files/14320032/Rttlog.txt)
Apple HomeKit quit letting me add accessory.     "Unable to Add Accessory.   Pairing with accessory failed"
attached file is the RTT logs that start before I entered the setup code 34970112332 in apple homekit.
ends with homekit message ... unable to add accessory

TomInIowa, 2024-02-24:
My old  SparkFun MGM240P quit working a week or so ago.  The reset button stuck down.  Have a new one now.
I added the board to my Apple HomeKit (two Apple TV 4K and one Apple HomePod) using the example Matter Temperature.
Eve app does **NOT** see it as a thread device!  User Brezensalze says his show up as thread devices. Mine does not.
Any idea on what I should look for?