SiliconLabsSoftware/arduino #17: Please Add Door Lock (0x000A)

ruddycanuck, 2024-02-14:
I am designing a latch style lock for my shed and I would like to use Arduino / Matter. I think this is a cool use of Matter / Matter over Thread if possible. There is a door lock example in connectedhomeip 
/connectedhomeip/examples/lock-app/

Minimal Viable Product:
ID: 0x000A
setLockState : Lock/Unlock
getDoorState: Door Open/Closed
readBattLevel: Battery Level
Tamper Alert

Secondary (Complex) Phases
Keypad 
RFID
Users

silabs-bozont, 2024-02-15:
Hello ruddycanuck,
_Matter Door Lock_ along with a few new devices is on our roadmap, so it will be available in one of the upcoming releases! Hang tight!

Callum6677, 2024-02-23:
> Hello ruddycanuck, _Matter Door Lock_ along with a few new devices is on our roadmap, so it will be available in one of the upcoming releases! Hang tight!

Is there any public version of this roadmap that can be shared (sanitised if need be, timelines not essential because I know myself how unrealistic it can be to give them). I have a few things I was interested in that aren't yet possible, but don't want to waste your time cluttering GitHub issues with things you already have planned

silabs-bozont, 2024-02-25:
Hi Callum6677,
I'll try to put something together on our goals, but feel free to ask!

lboue, 2024-05-08:
Not included in [Release 2.0.0](https://github.com/SiliconLabs/arduino/commit/65a739a859fa8b98e755f23f2b9c8010d25ca1e2) [Matter examples](https://github.com/SiliconLabs/arduino/tree/main/libraries/Matter/examples)

[image]

James-Smart, 2024-05-29:
Hi wanted to ask about this as well? How far off do you think it is? Deciding if its worth forking and adding. Cheers

lboue, 2024-05-29:
> Hi wanted to ask about this as well? How far off do you think it is? Deciding if its worth forking and adding. Cheers

@James-Smart I think you can make a fork because it takes several months between each new version.

James-Smart, 2024-05-31:
Excellent, thanks for the info

lboue, 2024-07-05:
Hello @silabs-bozont 
Any news about the next release?

silabs-bozont, 2024-08-01:
Hello @ruddycanuck,
Matter Door Lock is now available in 2.1.0.
See: https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_door_lock/matter_door_lock.ino

closed 2024-08-01 by silabs-bozont