SiliconLabsSoftware/arduino #18: Matter fan: Initializing OpenThread stack issue

lboue, 2024-02-15:
## Matter fan issue

I build the matter fan [matter_fan.ino](https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_fan/matter_fan.ino) example for [xG24 Explorer Kit](https://www.silabs.com/development-tools/wireless/efr32xg24-explorer-kit). 
After flashing program and bootloader and reboot the board it remains at the 'Initializing OpenThread stack' step:

```
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]Arduino Matter device starting
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]Init CHIP Stack
[00:00:00.000][silabs ]Initializing OpenThread stack
```

[image]

Can you help me correct this issue?

## Context
I tried the ligh example with Simplicity studio and it worked fine.

silabs-bozont, 2024-02-15:
Hello lboue,

Do you have anything on the Serial output? The RTT viewer likes to disconnect silently after the device resets, so manually reconnecting after a reset is advised - your device could be operating normally.

lboue, 2024-02-15:
> Hello lboue,
> 
> Do you have anything on the Serial output? The RTT viewer likes to disconnect silently after the device resets, so manually reconnecting after a reset is advised - your device could be operating normally.

As you can see from the screenshot, I've got nothing on the serial port apart from a strange character.

I get a different behavior with the other example files.
I did few resets and I have the same issue.

lboue, 2024-02-15:
Same issue with [matter_thermostat.ino](https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_thermostat/matter_thermostat.ino) with SparkFun ThingPlus Matter MGM240P:

```
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]Arduino Matter device starting
[00:00:00.000][silabs ]==================================================
[00:00:00.000][silabs ]Init CHIP Stack
[00:00:00.000][silabs ]Initializing OpenThread stack
```

lboue, 2024-02-15:
I think I've find the root cause. I tested and compared by compiling on another PC with Windows10.
Compiling changed nothing. However, I realized that the version of SimplicityCommander I was using was very old and the "Erase User Data" button was missing.

After using this button, the program started up correctly.
I'm going to test it with the other programs and see if it fixes my application startup problem.

lboue, 2024-02-16:
It worked with [matter_fan.ino](https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_fan/matter_fan.ino) too:

[image]

closed 2024-02-16 by lboue