SiliconLabsSoftware/arduino #19: [Feature request] Using onboard button to operate the device

lboue, 2024-02-16:
Hi,

Would it be possible to use the BTN0 and BTN1 buttons on the xG24 Explorer Kit board to operate the device in the following examples?
- matter_lightbulb.ino
- matter_fan.ino

[image]

silabs-bozont, 2024-02-16:
Hi Iboue,

The next release will contain a lightbulb example which is exactly the same as you described.
I can leak the code for it :)
```
/*
   Matter lightbulb with button example

   The example shows how to create a lightbulb with a controlling button using the Arduino Matter API.

   The example lets users control the onboard LED with Matter and also with the on-board button.
   State changes from the button and the Matter network are synchronized and the on-board LED is set accordingly.
   The device has to be commissioned to a Matter hub first.

   Compatible boards:
   - Arduino Nano Matter
   - SparkFun Thing Plus MGM240P
   - xG24 Explorer Kit
   - xG24 Dev Kit

   Author: Tamas Jozsi (Silicon Labs)
 */
#include <Matter.h>
#include <MatterLightbulb.h>

MatterLightbulb matter_bulb;

void handle_button_press();
volatile bool button_pressed = false;

void setup()
{
  Serial.begin(115200);
  Matter.begin();
  matter_bulb.begin();

  // Set up the onboard button
  #ifndef BTN_BUILTIN
  #define BTN_BUILTIN PA0
  #endif
  pinMode(BTN_BUILTIN, INPUT_PULLUP);
  attachInterrupt(BTN_BUILTIN, &handle_button_press, FALLING);

  // Set up the onboard LED
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  Serial.println("Matter lightbulb with button");

  if (!Matter.isDeviceCommissioned()) {
    Serial.println("Matter device is not commissioned");
    Serial.println("Commission it to your Matter hub with the manual pairing code or QR code");
    Serial.printf("Manual pairing code: %s\n", Matter.getManualPairingCode().c_str());
    Serial.printf("QR code URL: %s\n", Matter.getOnboardingQRCodeUrl().c_str());
  }
  while (!Matter.isDeviceCommissioned()) {
    delay(200);
  }

  if (!Matter.isDeviceConnected()) {
    Serial.println("Waiting for network connection...");
  }
  while (!Matter.isDeviceConnected()) {
    delay(200);
  }
  Serial.println("Device connected");
}

void loop()
{
  // If the physical button state changes - update the lightbulb's on/off state
  if (button_pressed) {
    button_pressed = false;
    // Invert the on/off state of the lightbulb
    bool bulb_state = matter_bulb.get_onoff();
    matter_bulb.set_onoff(!bulb_state);
  }

  // Get the current on/off state of the lightbulb
  static bool matter_lightbulb_last_state = false;
  bool matter_lightbulb_current_state = matter_bulb.get_onoff();

  // If the current state is ON and the previous was OFF - turn on the LED
  if (matter_lightbulb_current_state && !matter_lightbulb_last_state) {
    matter_lightbulb_last_state = matter_lightbulb_current_state;
    digitalWrite(LED_BUILTIN, HIGH);
    Serial.println("Bulb ON");
  }

  // If the current state is OFF and the previous was ON - turn off the LED
  if (!matter_lightbulb_current_state && matter_lightbulb_last_state) {
    matter_lightbulb_last_state = matter_lightbulb_current_state;
    digitalWrite(LED_BUILTIN, LOW);
    Serial.println("Bulb OFF");
  }
}

void handle_button_press()
{
  static uint32_t btn_last_press = 0;
  if (millis() < btn_last_press + 200) {
    return;
  }
  btn_last_press = millis();
  button_pressed = true;
}
```
Modifying this to use a Fan instead of a lightbulb is pretty easy - just change the Lightbulb class to a Fan class.

lboue, 2024-02-16:
Thanks for the quick reply, I'll give it a try.

closed 2024-02-16 by lboue

lboue, 2024-02-16:
Works fine 👍 
```
13:13:12.033 -> Matter lightbulb with button
13:13:12.600 -> Matter device is not commissioned
13:13:12.600 -> Commission it to your Matter hub with the manual pairing code or QR code
13:13:12.600 -> Manual pairing code: 34970112332
13:13:12.600 -> QR code URL: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A6FCJ142C00KA0648G00
13:14:00.666 -> Waiting for network connection...
13:14:02.073 -> Device connected
13:15:17.149 -> Bulb ON
13:15:20.079 -> Bulb OFF
13:15:25.673 -> Bulb ON
```