SiliconLabsSoftware/arduino #22: OneWire ... DS18B20 problem.  on my Silicon Labs MGM240P using Matter library

TomInIowa, 2024-02-16:
OneWire oneWire(ONE_WIRE_BUS);                                          //   libary installed by    DallasTemperature
  DallasTemperature sensorsDS18B20(&oneWire);                    //.  libary   DallasTemperature by Miles Burton

If I put these lines before setup(),  the sketch **hangs** and never gets to the setup function.
If I put these lines in the setup(), 

    sensorsDS18B20.requestTemperatures();
    Serial.print(sensorsDS18B20.getTempFByIndex(0));

returns       -196.60  about 1/2 of the time,  
returns the correct temperature from the DS18B20 sensor the rest of the time.

I tried different DS18B20 sensors. (that work on ESP32 sketch)

Trying to figure out how to use the debugger with the lines before the setup() function???

TomInIowa, 2024-02-17:
I am trying to use two DS18B20 temperature sensors in a Matter project on my SparkFun Thing Plus MGM240P board.
The library OneWire crashes the sketch on my board.  Attached is a small sketch.  If I don't comment out line 13 (OneWire oneWire(ONE_WIRE_BUS); ).  , the sketch crashes.   Bad Library??  MGM240P flaw?  

the attached sketch is just part of the Matter sketch... it shows my problem

[DS18B20_Problem.txt](https://github.com/SiliconLabs/arduino/files/14319527/DS18B20_Problem.txt)

silabs-bozont, 2024-02-19:
Hello TomInIowa,

If you want to see where the sketch crashes you can start debugging, then press the *pause* icon. You'll be presented with the code line where the instruction pointer is and the *call stack* will show you how did you get there.

I'm familiar with the issue you're having with the OneWire library. When you create an instance with
```
OneWire oneWire(C2);
```
the constructor for OneWire tries to initialize the GPIO from the init code - and that's a first class ticket to a crash 😢. The init code runs before `main()` and the GPIO clocks are not enabled at that point. So writing any GPIO register - like the OneWire library does by calling `pinMode` results in a crash. I'll take a look into making these early calls have no effect instead of causing a crash.

Anyway - to avoid this you can create the instance without specifying the pin and then call `begin()` in your setup like this:
```
OneWire oneWire;

void setup() {
  ...
  oneWire.begin(C2);
  ...
}
```
This will create a OneWire instance without initializing the GPIO. The GPIO init is done in `setup()` where the clocks have been enabled.

That's half of the way to getting it to work :)

The OneWire protocol requires strict microsecond timing - which usually cannot be achieved through the Arduino APIs. This is why each chip vendor has their [direct GPIO register access functions](https://github.com/PaulStoffregen/OneWire/blob/master/util/OneWire_direct_gpio.h) in the OneWire library. 

I've been working on support for our hardware - it's currently in alpha state, but I had success reading a _DS18B20_ with it. Once it's ready and well tested I'll submit a PR to the original OneWire library to make the support official.
You can test it if you'd like, here is the repo: https://github.com/silabs-bozont/OneWire
Just replace your current OneWire library with this and you're ready to go.
Let me know how it went!

TomInIowa, 2024-02-24:
Thank you!  It works.    It took awhile to find out.  My SparkFun board quit working... the reset button stuck down.