SiliconLabsSoftware/arduino #23: Force Static endpoints (non-bridge) for matter_lightbulb.ino

lboue, 2024-02-17:
Hi,

Is there any way for a device to be exposed as an Matter unbridged accessory?
With [matter_lightbulb.ino](https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_lightbulb/matter_lightbulb.ino) example containing only one bulb , is-it possible to use bridge mode with dynamic endpoints.

silabs-bozont, 2024-02-19:
Hello Iboue,

With the current version unfortunately it's not possible. Every device will also have an additional bridge endpoint as well.

The Matter library uses dynamic endpoints which are created during runtime. The Bridge endpoint was added as a kind of workaround because some ecosystems would not recognize dynamic endpoints without having a bridge.
We're currently exploring some solutions to have it removed and have regular endpoints instead of bridged ones, but till then the bridge is here to stay for a bit. The important thing is that in the future we expect to get rid of the Bridge (or make it optional) and have regular endpoints.

lboue, 2024-05-07:
Hello @silabs-bozont. Is this possible to use dynamic endpoints  with the new version 2.0.0 or will it come later?

silabs-bozont, 2024-05-07:
Hello @lboue - the library still uses the same method with the dynamic endpoints and the hub - no way yet to use static endpoints.

silabs-bozont, 2025-05-05:
Hello @lboue,
I'm closing this issue as static endpoints won't be supported on the Arduino Core. Code for the static endpoints is generated with the ZAP tool dynamically from a large config file before compilation - which we have no easy way and intention to integrate into the Arduino IDE.
The Arduino Matter Library uses dynamic endpoints by design - and we don't plan on changing this in the foreseeable future.
If you wish to use static endpoints - I'd advise you to look into developing with Simplicity Studio - which integrates the ZAP tool and uses static endpoints.
However if you wish to disable the bridge endpoint in the Matter library you can call `Matter.disableBridgeEndpoint()` - but this may result in wrong behavior on some ecosystems. Your endpoints will still be dynamic, but without the bridge device present.

closed 2025-05-05 by silabs-bozont