SiliconLabsSoftware/arduino #27: Some constants are not defined

rei-vilo, 2024-02-25:
I couldn't find some constants like &pi;, &half;&pi;, e, among others in [Arduino.h](https://github.com/SiliconLabs/arduino/blob/956de5de292f53d907c2746cacfb5a26de3e49fb/cores/silabs/Arduino.h).

```
#define PI 3.1415926535897932384626433832795
#define HALF_PI 1.5707963267948966192313216916398
#define TWO_PI 6.283185307179586476925286766559
#define DEG_TO_RAD 0.017453292519943295769236907684886
#define RAD_TO_DEG 57.295779513082320876798154814105
#define EULER 2.718281828459045235360287471352
```

Compilation fails with an error.

> sketch_feb25a.ino:3:15: error: 'PI' was not declared in this scope; did you mean 'SPI'?

silabs-bozont, 2024-02-25:
Hi! 
Thanks for bringing this oversight to my attention, the next release will have these defined!

silabs-bozont, 2024-05-07:
It's now fixed in 2.0.0

closed 2024-05-07 by silabs-bozont