SiliconLabsSoftware/arduino #28: [Feature request] Is it possible to use a deep sleep mode?

lboue, 2024-02-26:
Hello,

Is it possible to use a deep sleep mode with this arduino framework?

Regards

LorbusChris, 2024-05-09:
Since it's related: I've added an example for battery percent reporting via Matter here: https://github.com/SiliconLabs/arduino/pull/51

Looking forward to the low power lib support :)

miroslavpetrov, 2024-05-16:
@silabs-bozont Would you care to share an example on how to implement deep sleep using GSDK in Arduino?

silabs-bozont, 2024-05-29:
Hello @miroslavpetrov,
The current version already has the Power Manager SDK component in place, so you're able to harness it and go to a low power mode.
https://docs.silabs.com/gecko-platform/4.3/platform-service-power-manager-overview/
The easiest way to got to a low power mode is to block the user OS task (the core runs on FreeRTOS underneath).
You can block your task by waiting for a mutex/semaphore/queue - or also by calling `delay()`.
When the OS is running the `idle` task the Power Manager will automatically put the device in the lowest possible sleep mode.
It's also advised to turn off peripherals (Serial, Wire, SPI) to allow lower sleep modes.
Note, that if you're using a radio stack then it can (and will) periodically wake the MCU.

`ArduinoLowPower` library support is under development and should be available in the next version which will take care of these for you, but in the meantime you can achieve similar power savings with the tips above.

lboue, 2024-07-05:
> `ArduinoLowPower` library support is under development and should be available in the next version which will take care of these for you, but in the meantime you can achieve similar power savings with the tips above.

Hello @silabs-bozont 

Will you be releasing the new version soon with `ArduinoLowPower` support?

Regards

silabs-bozont, 2024-08-01:
Hello @lboue,
ArduinoLowPower support is now available with 2.1.0.
See: https://github.com/SiliconLabs/arduino/tree/main/libraries/ArduinoLowPower

closed 2024-08-01 by silabs-bozont

lboue, 2024-08-01:
Thanks @silabs-bozont