SiliconLabsSoftware/arduino #30: README incorrectly specifies location to add repo's URL

ericwertz, 2024-03-03:
In the README, in the Installation section, it says to go to "Settings" to splice-in the URL for the SiLabs repo to fetch from.  It should direct the user to "File->Preferences" within the Arduino IDE instead.

silabs-bozont, 2024-03-03:
Hi,
Thanks for the heads up! Looks like it's different on each OS.
Mine looks like this:
[image]
Will update the readme to reflect all platforms correctly.

ericwertz, 2024-03-03:
Yes.  I have bad karma, so I'm using Windows.  Didn't think to specify that.  Regards.

image.png (view on web)<https://github.com/SiliconLabs/arduino/assets/124884281/7a3f99b9-b904-4cdb-9ea0-d0fd9f87689f>

silabs-bozont, 2024-03-07:
No worries, Windows is not bad karma, it's totally fine :)

silabs-bozont, 2024-05-07:
Updated readme in 2.0.0

closed 2024-05-07 by silabs-bozont