SiliconLabsSoftware/arduino #31: Cannot get ezBLE.connected() == TRUE

Marce68, 2024-03-10:
I'm running ezBLE examples on a BGM220 Explorer kit.
The sketch gets stuck in the loop
while (!ezBLE.connected()) ;
I'm connecting to the BGM explorer kit using a smartphone running Apps for Bluetooth debugging like EFR Connect and nRF Connect.
Tested with bonding and without bonding, nothing changes, it looks like the board/sketch does not detect the connection.
Should I install a specific stack using Simplicity Studio?

silabs-bozont, 2024-03-12:
Hey Marce68!

Glad you're checking out **ezBLE**! It was initially designed for communicating between two Arduino boards over BLE, but you can make it work with your BGM220 Explorer Kit and your phone with **EFR Connect** as well.
The library expects the other side to have an ezBLE service and characteristic - otherwise it won't go into the connected state as it's not recognizing the other device as an ezBLE peer. You can set up your local GATT to have this characteristic - and once you do this you'll be able to send/receive data from your phone.

I'll walk you through the steps. First of course you'll need the **EFR connect** app. Go to the **Configure** tab on the bottom, then select **GATT Configurator** on the top.
[image]

You'll most likely have an empty list here, but go ahead and press the **Create New** button, then click the *edit* (pencil) icon on the newly created GATT server. Change the name to your preference, I named it ezBLE.

Add a **Service** with the name *ezBLE* and with the 128-bit UUID **de8a5aac-a99b-c315-0c80-60d4cbb5beef**. This is the service your Arduino device will look for after connecting.

Once you've added the service, go ahead and press the **Add Characteristic** button on the service you just created. Name it *ezBLE data* with the 128-bit UUID of **5b026510-4088-c297-46d8-be6c7367beef** - allow **read** and **write** - then save it.
It should look like this:
[image]

Go back once, and make sure your newly created GATT database **is enabled** (toggle is blue).

Now you can go to the **Scan** tab, look for your device and connect to it. Your Arduino device should be acting as an ezBLE server. Now when you connect to the device it should get past the initial state. You'll be able to send messages by writing to the ezBLE characteristic, or read messages by reading your own local ezBLE characteristic.

[image]

[image]

Let me know if this worked for you!

Marce68, 2024-03-15:
Thanks a lot for the detailed reply, it worked perfectly!
I didn't catch the required "simmetry", in this way it's really easy to use.
One more question.
In the past I used Bluetooth Xpress as an SPP over BLE and I have several projects still using it. Is it on the Arduino roadmap?

Thank you!

closed 2024-08-01 by silabs-bozont