SiliconLabsSoftware/arduino #32: Arduino IDE. SparkFun Thing Plus Matter - MGM240P no output in serial monitor

ihakov2, 2024-03-25:
Hi.
I have the issue with Arduino IDE
Arduino IDE.
 I uploaded the Matter lightbulb example to SparkFun Thing Plus Matter - MGM240P board. 
Burn bootloader using Simplicity Commander programmer, the bootloader file looks right: 
C:\Users\xxxx\AppData\Local\Arduino15\packages\SiliconLabs\hardware\silabs\1.0.0/bootloaders/sparkfun-thingplus-matter-bootloader-storage-internal-single-512k.hex

But nothing output in Serial monitor or Putty, I tried 'help, config, onboardingcode none,  Enter key' - nothing.

When I uploaded the code and bootloader using Simplicity IDE, I connected through the Putty and it was responding to commands. And was able to add device to Google Home app.

I prefer to use Arduino IDE. Please advise.
ildus

silabs-bozont, 2024-03-25:
Hello ihakov2,
You can also flash the bootloader from the IDE by selecting the programmer in the **Tools** menu, then selecting **Burn Bootloader**.
Could you please check that your board is alive by flashing the Blink example - and seeing if the LED blinks?
If that works then I'd advise to check a dead simple example which prints something on Serial - you could modify the blink example like this:
```
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  Serial.println(millis());
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}
```

It's also possible that you just simply miss the boot messages before you open your terminal. You could try opening your terminal program and pressing reset on the board after. The Arduino Matter variants doesn't not include the Matter CLI, so they won't react to any Serial data you send to them - the Serial port is handled by the user entirely.

Let me know if any of these fixed your issue!

ihakov2, 2024-03-26:
Thanks for your response. 
1. Yes, I burn bootloader in Arduino IDE and uploaded the code - nothing in Arduino's Serial Monitor.
    Could not connect Putty to serial port, looks like when the board connected to Arduino IDE it is not possible.
2. Tried your suggested modified Blink - no output in Serial Monitor, even the LED was not blinking. I remember when I got the board and loaded Blink, it was blinking.
3. Thought that the board is dead, tried again Simplicity IDE and was able to add to Google Home and control the Led from phone. Note: With uploaded code and bootloader in Simplicity IDE was able to enter commands: onboardingcodes none in Arduino IDE' serial monitor panel and board responded with the url and other info. So that means serial works. 
Confused, frustrated. No ideas.
PS: The LED CHG (charge) is permanently yellow. Is it right?

silabs-bozont, 2024-03-27:
That's actually good news that it works from Studio - means your board is not dead :) We'll get to the bottom of this! 
- What OS and what version of the Arduino IDE are you using?
- Please try to erase the board fully (with Simplicity Commander or from Studio) and burn the bootloader again from the Arduino IDE. Make sure you're selecting the correct variant - BLE and Matter have different bootloaders - so be sure you're on the correct variant before pressing _Burn Bootloader_. When you switch between Matter and BLE you have to re-burn the bootloader. (this will be eliminated in the next release)
- Are you using 115200 baud or something different? Please try with 115200 as it's the default.

Let me know if any of these helped!

ihakov2, 2024-03-27:
Thanks for the response.
1. Windows 10, Arduino 2.3.2
2. I tried that before. I did re-burn the bootloader in Arduino. This what I saw in the output when I burn it and it looks right: 
C:\Users\xxxx\AppData\Local\Arduino15\packages\SiliconLabs\hardware\silabs\1.0.0/bootloaders/sparkfun-thingplus-matter-bootloader-storage-internal-single-512k.hex
and then upload the Matter lightbulb example in Arduino.
 Did not fix the issue for Arduino. No output, not able to add to Google home. Even the simple blink example stop working.
So, saying that, Simplicity works, Arduino does not.

My goal is to use Arduino IDE. The SparkFun Thing Plus Matter - MGM240P board does not work in Arduino for me.
Returned the board to replace with new one. Maybe my board was defective.
Let you know once I get the new one.
Thanks.
ildus

ihakov2, 2024-03-30:
Hi silabs-bozont.

Good news. Got my replaced board today. 
Burn bootloader in Arduino IDE, flashed the code. Pressed the reset button and got the output in Arduino serial monitor. Added to Google Home. Worked. Tried another example - all good.
My goal accomplished. Ticket can be closed.
PS I did not tried the SiliconLabs.
Thanks.
ildus

silabs-bozont, 2025-04-08:
Hey @ihakov2,
Glad it worked out! Happy hacking!

closed 2025-04-08 by silabs-bozont