SiliconLabsSoftware/arduino #35: EEPROM library is missing

lyusupov, 2024-04-02:
Most of Arduino Cores on the market provide generic access or emulation of **EEPROM** subsystem.
The interface has become standard and is well explained here:
https://docs.arduino.cc/learn/built-in-libraries/eeprom/

Missing **EEPROM** feature makes the Silicon Labs Arduino Core to be less compatible with sketches and libraries available.

silabs-bozont, 2024-04-23:
Hello @lyusupov,

Thanks for your suggestion - yes, we currently don't support the official EEPROM library - but there are plans to support it in the future.
Thank you for your patience!

silabs-bozont, 2024-08-01:
Hello @lyusupov,
EEPROM library support is now available with 2.1.0.
See: https://github.com/SiliconLabs/arduino/tree/main/libraries/EEPROM