SiliconLabsSoftware/arduino #44: Not able to commission lightbulb_color example code

yogeshiggalore, 2024-04-12:
I followed the following steps for matter_lightbulb.

Erased the flash chip using Simplicity Studio. ( to erase previously stored commission data )
update bootloader code using burn bootloader.
programmed matter_lightbulb example code.
commissioned matter device using alexa mobile app and echo 4th gen.
Tested with alexa mobile app and echo, worked as expected.
for matter_lightbulb_color example

Followed step 1 and step 2.
programmed matter_lightbulb_color example code.
commissioned failed, in alexa app it is asking to select thread border router to commission but in the device serial print says "device commissioned".
Not able to understand the issue. repeated above mentioned steps multiple times. same result.

TomInIowa, 2024-04-13:
erase the chip ...    I've used their Commander app to do it.

silabs-bozont, 2025-04-08:
Closing due to inactivity - please reopen if the issue still persist on the latest release!

closed 2025-04-08 by silabs-bozont