SiliconLabsSoftware/arduino #49: [request] Add Matter 1.2 and Matter 1.3 device  type examples

lboue, 2024-05-08:
### Summary

Describe your feature request in a few lines

Could you add the new device types contained in the following specifications for the next version?

- Matter 1.2 Application Cluster Specification
- Matter 1.3 Application Cluster Specification

silabs-bozont, 2024-05-13:
Hello @lboue,
Our end goal is to support all possible device types, so they're definitely coming.
If I remember correctly CSA just released Matter 1.3 - which right now only means that the specification is available and the codebase implementing it is not yet. https://github.com/project-chip/connectedhomeip/releases
When the official Matter 1.3 codebase releases our Matter team will start integrating it into our Matter SDK - then the core will integrate the Silabs Matter SDK - so the 1.3 support may be a bit further down the line as of now.

lboue, 2024-06-18:
Now Matter 1.3 has been released for SiliconStudio:

https://github.com/SiliconLabs/matter_extension/releases/tag/v2.3.0

lboue, 2024-09-05:
Hi,

Is there any news about the new Matter 1.2 device types?

Regards

lboue, 2024-10-08:
Matter 1.2 was released almost a year ago. Any news on the integration of these types of devices?

Regards