SiliconLabsSoftware/arduino #50: [request] Support the option to enable oscillation of the fan head.

lboue, 2024-05-08:
### Summary

Support the option to enable oscillation of the fan head.

https://github.com/SiliconLabs/arduino/blob/65a739a859fa8b98e755f23f2b9c8010d25ca1e2/variants/xg24devkit/matter/matter_2.2.0/zzz_generated/app-common/app-common/zap-generated/cluster-enums.h#L2838

silabs-bozont, 2024-05-13:
Hello @lboue,
We'll definitely look into supporting these options in the next release.

lboue, 2024-07-05:
Hello @silabs-bozont 

Will you be releasing the new version soon?

Regards

silabs-bozont, 2025-07-31:
Hello @lboue,
It's on our roadmap - but unfortunately we haven't gotten to it yet.
We're gearing up for opening PRs for everyone, so if you'd like to contribute it in the future that'd also be great! :)