SiliconLabsSoftware/arduino #52: [request] Support for PlatformIO IDE

LorbusChris, 2024-05-16:
### Summary

I'd love to be able to use this Arduino core with PlatformIO.

I've done some initial work to create a PlatformIO framework builder here: https://github.com/platformio/platform-siliconlabsefm32/pull/8

Please feel free to re-use that, I'll likely not get around to finishing it myself.

silabs-bozont, 2024-05-17:
Hello @LorbusChris,
PlatformIO support is on our list of plans, so it'll be coming in the future!

balthz, 2024-12-30:
Do you have any update on this? I'm also interested in PlatformIO support for the Arduino Matter and excited to hear this is on you folks' list of plans!

wb8wka, 2026-10-03:
It's been 2 years; just checking in.  It would be spectacular to have this in PlatformIO/Visual Code Studio so I could use Claude Code with it.  It's doing astonishing things on a project I am doing with the ESP32 and I have no doubt it could so just as well with the Silabs Arduino build.

P.S. Getting an Arduino port to your WiFi chip would be an ESP32 killer, and expose this first in class low power WiFi chip to the masses

silabs-BelaV, 2026-10-05:
Thanks for following up, and apologies for the lack of an update.
Our previous comment reflected our plans at the time. PlatformIO remains a community-requested integration, but we currently do not have an official implementation or a committed timeline for it.
The community framework builder linked above could be a useful starting point for anyone interested in continuing the work. We would be happy to follow the progress and discuss it in this issue.

If you are using VS Code, I recommend checking out the OpenOCD extension, which can be used together with our SiSDK VS Code extension to work with Arduino and Seeed boards. Claude Code can support this workflow as well. Zephyr is another clear path for VS Code-based development.