SiliconLabsSoftware/arduino #55: [bug] <title>

oidebrett, 2024-05-23:
### Hardware

Arduino Nano Matter

### Core version

2.0.0

### Arduino IDE version

2.3.2

### Operating system

Windows 11

### Radio stack variant

Matter

### OpenThread Border Router device (if using Matter)

RPi

### Issue description

Shouldnt the OnOff cluster be on Endpoint 1 not Endpoint 3 in the Matter light example?

Is there a reason why the OnOff cluster for the Matter light example is on endpoint 3 rather than endpoint 1?
see
libraries/Matter/examples/matter_lightbulb/matter_lightbulb.ino

I am testing with the chip-tool and after pair I am trying to interact with the OnOff cluster. In most other Matter development platforms that primary device type clusters are on endpoint 1. But i have found that any matter example that I have tried from this repo the primary clusters are on Endpoint 3.

Am I missing something?

lboue, 2024-05-25:
@oidebrett 
You can use [SimplicityStudio ](https://www.silabs.com/developers/simplicity-studio) as a workaround. The examples do not include a bridge. It works correctly in this case.

silabs-bozont, 2024-05-29:
Hello @oidebrett,
This is by design - we use dynamic endpoints - so a bridge device was added on endpoint 2 to ensure compatibility with most ecosystems. Also, you can parametrize your chip-tool commands to look for the device on endpoint 3 and be able to interact.

oidebrett, 2024-05-29:
Thanks for your reply. I am just starting my journey into Arduino nano matter development and I am using the Arduino IDE. 

I have experience of using the ESp32 and I can write code using the Esp-idf to write code that implements endpoints, clusters etc. 

Is there any documentation on how I could use the Arduino IDE to get a lower level of access to the matter data model and core functionality? For example if I don't want to use a bridge on endpoint 2 could I implement my own endpoints and clusters?

For example with ESP-idf I can use write code at this level

https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#developing-your-product

Sorry to be asking this in the repo. Is there a better place to get this information that is regularly responded to?

Many thanks

silabs-bozont, 2024-06-17:
Hello @oidebrett,

No worries, you're in the right place for asking questions. The bridge endpoint was introduced to ensure compatibility for the bridges/dynamic endpoints. In the future versions we'll introduce a way to turn it off, however in the meantime you can disable it with:
`emberAfEndpointEnableDisable(emberAfEndpointFromIndex(1), false);`

If you want to go a bit lower  - take a look at the source of a device in the Matter library (like [this](https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/src/MatterTemperature.cpp) and [this](https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/src/devices/DeviceTempSensor.cpp)) - those directly access the Matter SDK. You can create your own custom endpoints/devices in the same fashion.

closed 2025-05-05 by silabs-bozont