SiliconLabsSoftware/arduino #56: [request] Add fast GPIO for WS2812 RGB LED

rei-vilo, 2024-06-05:
### Summary

The  [ezWS2812](https://github.com/SiliconLabs/arduino/tree/main/libraries/ezWS2812) library relies on SPI to drive the WS2812. Problem is, only on signal is (MOSI) but two others (MISO, SCK) are required.

The [Adafruit_NeoPixel](https://github.com/adafruit/Adafruit_NeoPixel) library does not seem to support SiLabs MCUs. 

The request is to use GPIO to drive the WS2912, either by adding support for SiLabs MCU to Adafruit_NeoPixel, or by implementing fast GPIO.

Thank you!

rei-vilo, 2024-06-13:
Arduino seems to provide direct access to Gecko-SDK.

[WS2813.zip](https://github.com/user-attachments/files/15815012/WS2813.zip) includes the code developed with Simplicity Studio with `GPIO_PinOutSet()` and `GPIO_PinOutClear()`.

It has been tested agasint the BG22 and BGM220P Explorer Kits and Arduino Nano Matter board. Faster MCUs may require an adaptation.

silabs-bozont, 2024-06-17:
Hello @rei-vilo,

Yes, you have full access to the underlying GSDK :)
A GPIO variant of ezWS2812 is in the works an will be coming in an upcoming release!

rei-vilo, 2024-06-17:
Great!

rei-vilo, 2024-12-01:
Release 2.2.0 works fine, thank you!

closed 2024-12-01 by rei-vilo