SiliconLabsSoftware/arduino #57: [request] Commission more than one device

metehoca, 2024-06-06:
### Summary

I have Nano Matter and EFR32xG24 Explorer Kit. I want to use them at the same time. But looks like Arduino Core gives them same commissioning ID and Google Nest Hub refuses to add second one.

There should be a way to change commissioning ID at the code.

silabs-bozont, 2024-06-17:
Hello @metehoca,

Thank you for bringing this to our attention. I've also noticed this restriction being introduced on my Google devices as well.
Unfortunately right now there's no easy way to change between test certificates, however we offer a provisioning tool which can be used to change your setup payload.
https://github.com/SiliconLabs/matter/tree/release_2.3.0-1.3/provision
It's not compatible with the Nano Matter right now - but you should be able to change your setup payload on your xG24 Explorer Kit.
If this restriction remains on the Google hubs we'll have to introduce a way to pair multiple devices easily like before.

silabs-bozont, 2024-08-01:
Hello @metehoca,
The provisioning script is now a part of the Core - you can change your device identities with it to allow you to commission more devices.
See: https://github.com/SiliconLabs/arduino/tree/main/extra/arduino_matter_provision

closed 2024-08-01 by silabs-bozont