SiliconLabsSoftware/arduino #59: [bug] Issue with debugging against Arduino Nano Matter with core 2.0.0

rei-vilo, 2024-06-10:
### Hardware

Arduino Nano Matter

### Core version

2.0.0

### Arduino IDE version

2.3.2

### Operating system

Debian 12 Gnome

### Radio stack variant

No radio

### Issue description

Follows #4 but with Arduino Nano Matter on core 2.0.0.

The Arduino IDE fails to launch `openocd`. However, the same command works fine on terminal window.

Could the  AppImage suffers from an issue similar to FlatPack as in https://github.com/arduino/arduino-ide/issues/2115

Thank you!

### Serial output

Waiting for gdb server to start...[2024-06-10T21:19:40.774Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/home/reivilo/.arduino15/packages/SiliconLabs/tools/openocd/0.12.0-arduino1-static/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/reivilo/Projets/Arduino/Blink_debug_Nano_Matter -f /tmp/.mount_Arduin6YEOpv/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/efm32s2_g23.cfg
[2024-06-10T21:19:40.807Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

### Minimal reproducer code

Blink example

facchinm, 2024-06-12:
@silabs-bozont @rei-vilo the output I'm getting from the IDE is
```

rei-vilo, 2024-06-12:
Sure, the output from the IDE is 

```
Waiting for gdb server to start...[2024-06-12T11:31:29.365Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/home/reivilo/.arduino15/packages/SiliconLabs/tools/openocd/0.12.0-arduino1-static/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/reivilo/Projets/Arduino/Blink_copy_20240109222257 -f /tmp/.mount_Arduin4TnFKr/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/efm32s2_g23.cfg
[2024-06-12T11:31:29.389Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

Adding the `debug_custom.json` file from the procedure [Debugging on the Arduino Nano Matter](https://github.com/SiliconLabs/arduino?tab=readme-ov-file#debugger-configuration-file-1) throws the same error message.

```
Waiting for gdb server to start...[2024-06-12T11:48:30.744Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/home/reivilo/.arduino15/packages/SiliconLabs/tools/openocd/0.12.0-arduino1-static/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/reivilo/Projets/Arduino/Blink_copy_20240109222257 -f /tmp/.mount_ArduinvUiANZ/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/efm32s2_g23.cfg
[2024-06-12T11:48:30.788Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

Trying a second time raises this error message. Is there a utility still running in the background?

[image]

However, the same command from the terminal succeeds.

```
$ /home/reivilo/.arduino15/packages/SiliconLabs/tools/openocd/0.12.0-arduino1-static/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/reivilo/Projets/Arduino/Blink_copy_20240109222257 -f /tmp/.mount_ArduinvUiANZ/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/efm32s2_g23.cfg
Open On-Chip Debugger 0.12.0+dev-01514-g21fa2de70 (2024-02-07-19:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
efm32s2_dci_read_se_status
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2341:0x0072, serial=EE54D8AC
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = EE54D8AC
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x6ba02477
Info : [efm32s2.cpu] Cortex-M33 r0p4 processor detected
Info : [efm32s2.cpu] target has 8 breakpoints, 4 watchpoints
Info : [efm32s2.cpu] Examination succeed
Info : starting gdb server for efm32s2.cpu on 50000
Info : Listening on port 50000 for gdb connections
```

Environment

```
$ uname -a
Linux NUC-12 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux
```