SiliconLabsSoftware/arduino #60: [bug] matter_fan.ino: after a certain period of time, the fan device returns to Off state by itself, without having to send a command to do so.

lboue, 2024-06-15:
### Hardware

Silicon Labs xG24 Explorer Kit BLE Matter

### Core version

2.0.0

### Arduino IDE version

2.3.2

### Operating system

Windows 11

### Radio stack variant

Matter

### OpenThread Border Router device (if using Matter)

Apple TV 4K

### Issue description

Detailed description of the issue

Sketch: [matter_fan.ino](https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_fan/matter_fan.ino)

After a certain period of time, the fan device returns to Off state by itself, without having to send a command to do so.

silabs-bozont, 2024-06-17:
Hello @lboue,

Can you please post your Serial log from the device?
Do you see a `Fan OFF` message when the fan turns back off, or is it happening without it being logged?
If it's possible to get and RTT log would also be useful.

lboue, 2024-06-17:
Thank you for your reply. I will collect the requested logs.

lboue, 2024-07-04:
Here are the requested logs:
[SparkFun Thing Plus Matter_Jlink_RTT.log](https://github.com/user-attachments/files/16102771/SparkFun.Thing.Plus.Matter_Jlink_RTT.log)
[SparkFun Thing Plus Matter_Serial.log](https://github.com/user-attachments/files/16102772/SparkFun.Thing.Plus.Matter_Serial.log)

closed 2024-08-27 by lboue