SiliconLabsSoftware/arduino #62: [request] Replace polling by event in Matter examples

rei-vilo, 2024-06-17:
### Summary

In many examples, delay before two consecutive data acquisition is performed by polling.

https://github.com/SiliconLabs/arduino/blob/65a739a859fa8b98e755f23f2b9c8010d25ca1e2/libraries/Matter/examples/matter_sensor_multiple/matter_sensor_multiple.ino#L61-L67

This might not be especially low-power.

How to use instead an event raised by a timer? As SiLabs core for Arduino seems to rely on FreeRTOS, the RTOS elements should be already available.

Thank you!

silabs-bozont, 2024-06-17:
Hello @rei-vilo,

Yes, the examples use polling - so they're might not be that power efficient.
We'll look into introducing event callbacks to the library so you can block your user task and wait for Matter events to happen - it's a good idea.

silabs-bozont, 2024-06-17:
Hello @rei-vilo,

I just realized what you need here is to replace the busy-looping with a simple `delay()` - that'll block the user task for the provided amount of time allowing the device to enter a low-power mode and also run the radio stack if necessary. We'll also introduce support for `ArduinoLowPower` to allow more control.

What I was thinking about is the ability to register a callback for your devices which is called when some property changes. This would allow you to block your user task and go low power until something changes - then only react to it once without constantly polling for changes. This will be very useful, so it'll be definitely coming.

silabs-bozont, 2025-04-08:
Hey @jan-xyz,

Calling `LowPower.deepSleep(30000)` shuts down the whole MCU including all the radios, so the device will be offline while it's sleeping - and it'll have to connect back up to the network when waking up.
Calling `delay()` will yield the user task for the given amount of time while allowing the radio and stacks to continue to work in a lower power state. It'll consume more power than deep sleeping, but much less than usual.

The most energy effective method is to use the callbacks of the Matter library - with which the device can be in a lower power mode and instantly react to changes on the Matter device.
Here's an example on how to use callbacks with the Matter library: https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_lightbulb_callback/matter_lightbulb_callback.ino

@rei-vilo the callbacks example also gives a solution to your issue, please take a look! You can also create more FreeRTOS tasks to handle you sensor inputs in parallel.

closed 2025-04-08 by silabs-bozont