SiliconLabsSoftware/arduino #65: [request] Support for Push Messages

rei-vilo, 2024-06-24:
### Summary

In blog [Matter 1.3 is Here - What it Means for Smart Home Adoption](https://www.silabs.com/blog/matter-1-3-is-here-what-it-means-for-smart-home-adoption), section **Matter 1.3 Brings Improvements to Existing Devices** mentions Push Messages and Dialogs. 

> Other areas of the release include improvements to existing devices. Casting Video Players, think TVs, adds significant updates. However, for me, one feature stood out, and that is Push Messages and Dialogs. This means Matter devices can now send notifications to the TV. In my house, I can get notifications over my voice assistants, however, I require hearing aids, which I seldom wear in the house. So, I struggle to comprehend audio message since I don’t have context to figure it out. However, the ability to display the message on my TV is a great feature for me.

Does the Matter library manages this new Push Messages and Dialogs type?

Thank you and best regards.

rei-vilo, 2024-06-26:
Might be related to #49

silabs-bozont, 2025-05-05:
Hello @rei-vilo,
The newest release will update the Matter standard used to 1.4 which will include the possibility to implement this.
The `Messages` cluster is the part of the `Video Player` and the `Casting Video Player / Client` devices - which is not yet supported by any of the mainstream hubs.
Right now we don't plan on introducing these types of devices - as they could only be controlled from a Raspberry Pi controller.

rei-vilo, 2025-05-06:
Thank you for the update.

closed 2025-05-06 by rei-vilo