SiliconLabsSoftware/arduino #66: Difficulty Integrating Multiple Switches with Different Press Types in Matter

tiziano149, 2024-06-29:
### Hardware

Arduino Nano Mater

### Core version

2.0.0

### Arduino IDE version

2.3.2

### Operating system

Mac os Sonoma

### Radio stack variant

Matter

### Issue description

I’m working on a project using the Arduino Nano Matter board with the Matter 2.0.0 library. My goal is to integrate four switches into a single accessory in Apple HomeKit. Each switch should support different press types, including short press, long press, and multi-press.

I’m encountering challenges where each switch appears as a separate device, and I’m struggling to implement the various press types using the MatterSwitch class. Methods like report_event or triggerEvent don’t seem to be available.

I would appreciate any guidance or examples on how to consolidate these switches into one accessory and properly handle different button press types within the existing library framework. Thank you for any assistance you can provide!

lboue, 2024-08-07:
I'd also like to make the same device.

leonardocavagnis, 2024-09-05:
The `MatterSwitch` class doesn't natively support multiple press types such as short, long, and multi-press. 

A possible approach could be to use the "Operational State Cluster", where you define a unique state for each switch condition (e.g., long press on switch 1, short press on switch 1, multi-press on switches 1 & 2, etc.).
Unfortunately, this cluster is currently not implemented in the core.

[image]
_(Matter Application Cluster Specification 1.2)_

A quick and dirty solution could be to use one of the "Measurement and Sensing" clusters (such as Humidity, Thermostat, etc.) and assign a numeric value for each state. I understand this isn’t an elegant solution, but it’s the only workaround that comes to mind. :)

JakobGlieder, 2024-12-13:
> The `MatterSwitch` class doesn't natively support multiple press types such as short, long, and multi-press.
>
> A possible approach could be to use the "Operational State Cluster", where you define a [...]

By this you mean the librarys class does not support it or the Matter Standard? because i have a button like this at home, and also wanted to create something simmilar.

[image]

JakobGlieder, 2024-12-13:
There should be specifications at least for long and multi press, since they are even mentioned in the button .h file

lboue, 2025-10-26:
Only SinglePress is supported here (NumberOfPositions=2):
https://github.com/SiliconLabs/arduino/blob/ddc757e9612e9c053c360467dac46c492d4e3933/libraries/Matter/src/MatterSwitch.cpp#L135-L137