SiliconLabsSoftware/arduino #67: [bug] SpeedMax attribute missing for Fan device type

lboue, 2024-07-04:
### Hardware

Silicon Labs xG24 Explorer Kit

### Core version

2.0.0

### Arduino IDE version

2.3.2

### Operating system

Windows 11

### Radio stack variant

Matter

### OpenThread Border Router device (if using Matter)

Apple TV 4K

### Issue description

SpeedMax attribute missing for Fan device type

### Minimal reproducer code

[matter_fan.ino](https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_fan/matter_fan.ino)

lboue, 2024-07-04:
Check my PR #68 for this.

silabs-bozont, 2024-07-29:
Hello @lboue,
Thank you for the finding and the PR. Unfortunately we don't have the community contributions set up yet - is it okay with you if I fix it in the upstream without your PR?

lboue, 2024-07-29:
Yes, no problem with that

closed 2024-08-01 by silabs-bozont

lboue, 2024-08-02:
Thanks @silabs-bozont for this fix.