SiliconLabsSoftware/arduino #70: [request] getOnboardingQRCodeUrl() with leading `MT:` instead of `MT%3A`.

rei-vilo, 2024-07-08:
### Summary

Is there a dedicated function to get the QR-code with leading `MT:` instead of `MT%3A`. 

Currently, `getOnboardingQRCodeUrl()` requires some added code.

```cpp
    // Get code URL
    // Remove http address
    char * buffer = strstr(Matter.getOnboardingQRCodeUrl().c_str(), "MT%3A") + 5 - 3;

    // Replace MT%3A by MT:
    buffer[0] = 'M';
    buffer[1] = 'T';
    buffer[2] = ':';

    displayQR(buffer);
 ```

Thank you!

silabs-bozont, 2024-07-10:
Hello @rei-vilo,
The next version will have an API for getting the onboarding payload - which can be used as-is for QR code generation.

silabs-bozont, 2024-08-01:
Hello @rei-vilo,
In the new 2.1.0 release you can call `Matter.getOnboardingQRCodePayload();` to get the raw onboarding payload.

closed 2024-08-01 by silabs-bozont