SiliconLabsSoftware/arduino #71: [request] Provide example for decommissioning

rei-vilo, 2024-07-10:
### Summary

Arduino provides an example for decommissioning at [Nano Matter User Manual > Device Decommissioning](https://docs.arduino.cc/tutorials/nano-matter/user-manual/#device-decommissioning)

However, this procedure fails to perform a full decommissioning. However, I need to perform two more steps: 

1. Flash the bootloader, and 
2. Remove the devices from the Home app.

The request includes two parts: 

1. a better clean-up function of the Matter device, and 
2. an exhaustive procedure for the border router and the controller.

Thank you and best regards, 

--- 

Environment:

- Arduino Nano Matter and XG24 Explorer Kit
- Appel TV 4K as border router
- Apple Home app on iPhone

silabs-bozont, 2024-07-10:
Hello @rei-vilo,
The next version will have a proper decommissioning API with an example.

silabs-bozont, 2024-08-01:
Hello @rei-vilo,
The new decommissioning API is now available in 2.1.0.
See: https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_decommission/matter_decommission.ino

closed 2024-08-01 by silabs-bozont