SiliconLabsSoftware/arduino #73: [bug] Serial interface locks up on Arduino Nano Matter on Mac

technoblogy, 2024-07-11:
### Hardware

Arduino Nano Matter

### Core version

2.0.0

### Arduino IDE version

2.3.2

### Operating system

macOS Catalina 10.15.7

### Radio stack variant

Matter

### OpenThread Border Router device (if using Matter)

N/A

### Issue description

In some situations the Serial interface locks up on the Arduino Nano Matter making further serial communication impossible until the board is reset.

I can demonstrate this in two different situations. These may have the same cause, or there may be two different problems:

### Entering text into the Serial Monitor causes the Serial to hang up after about 384 characters.

Run the following sketch:
```
void setup() {  
  Serial.begin(9600);
}

int Count = 0;

void loop (void) {
  if (Serial.available()) {
    Serial.read(); Count++;
    if (Count % 64 == 0) Serial.println(Count);
  }
}
```
Then cut and paste all the following text (total 512 characters) into the Serial Monitor input field, and press Return:

$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ

The Serial Monitor prints:

64
128
192
256
320
384

ie it hangs up after 384 characters and doesn't reach 512.

Once it hangs up the Serial is totally unresponsive, and the only solution seems to be to reset the processor.

This test works fine on all other Arduino boards I've tested.

### Entering a string of exactly 63 characters into the Serial monitor causes it to lock up.

Install this program:
```
void setup() { 
Serial.begin(9600);
}

void loop (void) {
if (Serial.available()) Serial.write(Serial.read());
}
```
Repeatedly copy and paste this string into the Serial Monitor, and press Return:

0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ&

After the first attempt the Serial Monitor locks up and becomes unresponsive.

I have verified that the issue doesn't occur on Windows 10.

I have previously reported a similar issue on the RP2040: https://github.com/arduino/ArduinoCore-mbed/issues/193

leonardocavagnis, 2024-07-11:
Dear @technoblogy,
thanks for report!
Do you encounter the same issue when using an external serial terminal (as `screen`,  CoolTerm, ...)?

leonardocavagnis, 2024-07-15:
I tested on my macOS Sonoma 14.5 and I encountered the same issue.
Unlike you,  it hangs up after the first 64 chars. After many attempts, I found out that issue occurs only if you send exactly 64 chars or a multiple of it (128, 192, ...) and only on macOS.

Thanks for check it and for your detailed report!
I'm working on resolving it.

technoblogy, 2024-07-29:
@leonardocavagnis any news about this?

I'd be happy to test a preliminary fix when you have one available.

leonardocavagnis, 2024-08-02:
@technoblogy Unfortunately, it seems a bug of MacOS system. I'm investigating on it, I will be back to you if I find out something.

technoblogy, 2024-08-02:
> Unfortunately, it seems a bug of MacOS system.

I don't see how that can be possible. I've tested my application with a variety of other boards and the problem doesn't occur with any of them. For examples of the boards that work see:

[uLisp - Performance](http://www.ulisp.com/show?36M3)

technoblogy, 2024-08-15:
@leonardocavagnis Any success with solving this? I'd love to make my Lisp system, uLisp, available on the Arduino Nano Matter, and this is the only remaining issue I need to resolve.

technoblogy, 2025-05-02:
@leonardocavagnis Any progress with fixing this?

technoblogy, 2025-05-03:
@leonardocavagnis I have now performed the following tests:

* Installed my test program using the SiliconLabs Arduino core on a Mac.

* Confirmed that using the Serial Monitor running on the Arduino IDE (2.3.2) on a Mac, it hangs up as described in this post.

* Confirmed that if I test it using the Serial Monitor running on the Arduino IDE (2.3.2) on Windows it does _not_ hang up.

However, I don't think this confirms that "it seems a bug of MacOS system", because as already mentioned above there are many microcontroller boards, including most Arduino boards, that work fine using the Serial Monitor on both Mac and Windows.

It seems more likely that there _is_ a difference in the Serial Monitor between Mac and Windows, but that it's possible to write the serial code to take account of this, and so far the SiliconLabs Arduino core doesn't do this.

leonardocavagnis, 2025-05-06:
@technoblogy 
Thanks so much for your patience and for running all those tests 🙏

You're absolutely right: this isn’t a general macOS issue, I explained it poorly. The problem seems to be related specifically to the SAMD11 chip used on some boards, like the Arduino Nano Matter, when running on macOS. Unfortunately, we haven’t found a definitive solution yet 😔

The SAMD11 is a small microcontroller used as a USB-to-Serial converter on the Nano Matter (and other Arduino boards).

I’ll keep you posted as soon as we have any updates!

technoblogy, 2025-05-06:
> I’ll keep you posted as soon as we have any updates!

Thanks!

technoblogy, 2025-05-07:
@leonardocavagnis I just remembered that the Arduino Nano Every uses the same SAMD11 USB-to-Serial converter. I have one of those boards so I just tried the test on that, and it works without problems.

So perhaps the USB-to-Serial code for that board will give a clue?

silabs-bozont, 2026-10-02:
Hey @technoblogy,
I managed to find the root cause of the issue and implemented a fix for it in the SAMD11's `free-dap` firmware.
macOS could send a zero length packet which stranded the SAMD11 firmware in a state where it wasn't listening to incoming packets.
You can find the fix for the issue with some other improvements on my branch here: https://github.com/silabs-bozont/free-dap/tree/nano_matter_samd11
Flashing a new SAMD11 firmware to your Nano might be a bit challenging, but definitely possible - you'll need an SWD compatible programmer and a way to interface with the SWD pads of the SAMD11 on the bottom of the Nano.
You can find the exact SWD pads here: https://docs.arduino.cc/resources/pinouts/ABX00112-full-pinout.pdf

technoblogy, 2026-10-02:
Thanks for investigating this after all this time!