SiliconLabsSoftware/arduino #74: [bug] <title>

ericwertz, 2024-07-11:
### Hardware

xG2x dev kits

### Core version

2.x

### Arduino IDE version

2.x

### Operating system

any

### Radio stack variant

BLE

### Issue description

ble_blinky and ble_hid_keyboard examples both contain the data-sharing-between-ISR-and-main bug where the shared data isn't declared volatile.  The code will only work as long as the optimizer is turned down/off and the globals' re-loads into registers aren't optimized away.

silabs-bozont, 2024-07-18:
Hello @ericwertz - please fill in the missing parts of the issue template - especially the title.
Thank you!

silabs-bozont, 2024-08-01:
Hello @ericwertz,
It has been fixed in 2.1.0 - thank you for the report!

closed 2024-08-01 by silabs-bozont