SiliconLabsSoftware/arduino #76: [bug] <Connecting Nano Matter to Google Home using an iPhone>

lrodriguez4999, 2024-08-02:
### Hardware

Arduino Nano Matter

### Core version

2.0.0

### Arduino IDE version

2.3.2

### Operating system

Windows 11

### Radio stack variant

Matter

### OpenThread Border Router device (if using Matter)

Google Nest Hub (2nd Gen)

### Issue description

I have an issue when connecting the Nano Matter with the Google Home Nest Hub (2nd Gen) with an iPhone. I have tried it with an Android phone and has worked well, however with the apple phone, it tries to add it through the iOS first and then Google Home. The result is that it fails all the time while the Nano is waiting for Matter device discovery.
[image]

### Serial output

[image]

silabs-bozont, 2024-08-22:
Hello @lrodriguez4999,
Unfortunately this is a known issue - I also encountered it - could be an interoperability issue between the Google Nest Hub and the iPhone. We'll do some further investigation to narrow down the cause.
In the meantime you can work this around by using an Android phone.

tiziano149, 2024-09-01:
Similar issue here with homekit!
Device not recognized even though it shows up.

leonardocavagnis, 2025-01-08:
Hey, recent iOS update should fix this issue!
Give it a spin and let me know @tiziano149 @lrodriguez4999

silabs-bozont, 2025-04-07:
Hey @lrodriguez4999 @tiziano149,
I can also confirm what @leonardocavagnis wrote - pairing to Google works with the latest iOS.
My setup is: iPhone 16 (iOS 18.4), Google Nest Hub Gen2
I'm closing this issue - we can reopen it for discussion if the issue resurfaces.

closed 2025-04-07 by silabs-bozont