SiliconLabsSoftware/arduino #77: [request] <New device Types Matter 1.3>

tiziano149, 2024-08-02:
### Summary

Hi, I'd like to suggest including the new device types from Matter 1.3 into the Silicon Labs Arduino Nano Matter library. This update would allow developers to integrate a wider variety of devices. Here’s the full list of new device types:

Utility Device Types:

    Root Node
    Power Source
    OTA Requestor
    OTA Provider
    Bridged Node
    Electrical Sensor
    Device Energy Management

Application Device Types:

Lighting:

    On/Off Light
    Dimmable Light
    Color Temperature Light
    Extended Color Light

Smart Plugs/Outlets and Other Actuators:

    On/Off Plug in Unit
    Dimmable Plug in Unit
    Pump
    Water Valve

Switches and Controls:

    On/Off Light Switch
    Dimmer Switch
    Color Dimmer Switch
    Control Bridge
    Pump Controller
    Generic Switch

Sensors:

    Contact Sensor
    Light Sensor
    Occupancy Sensor
    Temperature Sensor
    Pressure Sensor
    Flow Sensor
    Humidity Sensor
    On/Off Sensor
    Smoke CO Alarm
    Water Freeze Detector
    Water Leak Detector
    Rain Sensor

Closures:

    Door Lock
    Door Lock Controller
    Window Covering
    Window Covering Controller

HVAC:

    Heating/Cooling Unit
    Thermostat
    Fan
    Air Purifier
    Air Quality Sensor

Media:

    Basic Video Player
    Casting Video Player
    Speaker
    Content App
    Casting Video Client
    Video Remote Control

Robotic Devices:

    Robotic Vacuum Cleaner

Appliances:

    Refrigerator
    Temperature Controlled Cabinet
    Room Air Conditioner
    Laundry Washer
    Dishwasher
    Laundry Dryer
    Cook Surface
    Cooktop
    Oven
    Extractor Hood
    Microwave Oven

Energy Devices:

    EVSE

Adding these would make the library more versatile and useful. More details can be found in the [Matter Device Types documentation](https://handbook.buildwithmatter.com/howitworks/devicetypes/).

Thank you for considering this suggestion!

Tiziano

silabs-bozont, 2024-08-22:
Hello @tiziano149,
Thank you for your request. Our end goal is the same as your request - to support all Matter device types - and we'll keep adding support for new devices in each release.
Matter 1.3 [is not out yet](https://github.com/project-chip/connectedhomeip/milestone/65) officially, but we'll definitely be integrating it once it becomes widely available.

rei-vilo, 2024-10-17:
Also Push Messages and Dialogs from Matter 1.3 as in #65

loozhengyuan, 2024-12-11:
Looks like [Matter 1.4 is released](https://github.com/project-chip/connectedhomeip/tree/v1.4-branch) as well, will be good if we can include them 🙇

loozhengyuan, 2024-12-11:
@silabs-bozont Looks like [Matter 1.4 is released](https://github.com/project-chip/connectedhomeip/tree/v1.4-branch) as well, will be nice if we can have the updated device types in the upcoming releases. 🙇

tiziano149, 2024-12-11:
> @silabs-bozont Looks like [Matter 1.4 is released](https://github.com/project-chip/connectedhomeip/tree/v1.4-branch) as well, will be nice if we can have the updated device types in the upcoming releases. 🙇

[image]

Seems like it's not though...
A bit confused about the situation. Companies are releasing Matter 1.4 products but it's not out yet...

lboue, 2024-12-11:
I think there's a problem with this project because there are products certified to Matter 1.3 specifications but the [1.3 milestone](https://github.com/project-chip/connectedhomeip/milestone/65) is not yet complete.

[image]

It doesn't make sense.

rei-vilo, 2024-12-11:
No need to wait for 1.3 for the Matter device identification #105 feature.

lboue, 2024-12-11:
v1.4 has been released:

https://github.com/project-chip/connectedhomeip/releases/tag/v1.4.0.0

tiziano149, 2024-12-11:
> v1.4 has been released:
> 
> https://github.com/project-chip/connectedhomeip/releases/tag/v1.4.0.0

Cool! Interesting haha, it's been released 55 minutes ago.
Coincidence or was it just not updated yet?

tiziano149, 2024-12-11:
Now that it's here, back to the original topic:

It would be great to have this soon so we can keep up with prototyping as soon as the Matter Protocol is being updated.

lboue, 2025-02-03:
Still no progress on this?

tiziano149, 2025-06-24:
An update is really needed! My company invested in quite a few Arduino Nano Matter boards.
The development going stale so quickly is unacceptable.
Please update the platform.

lboue, 2026-01-16:
Why don't you publish an update? 
It's been almost six months since the last version.