SiliconLabsSoftware/arduino #80: Duplicate of pins assignment for BGM220

lyusupov, 2024-08-03:
### Hardware

BGM220

### Core version

2.1.0

### Arduino IDE version

any

### Operating system

any

### Radio stack variant

No radio

### OpenThread Border Router device (if using Matter)

none

### Issue description

both **PC7** and **PA4** are assigned to 2 different Arduino pin numbers

[image]

silabs-bozont, 2024-08-22:
Hello @lyusupov,
Eagle eyes, thanks for noticing. Redundant mappings (15, 16) will be removed in the next release.

lyusupov, 2024-08-22:
@silabs-bozont 

Thank you!

One more issue to talk about - when **32768 Hz XTAL is not connected** to PD0 and PD1 pins  - 
- delay() and delayMicroseconds()
- millis() and micros()
are **operating approx. 100-110 times slower than they should**.

I found out that the Core relies on the 32.768 Hz XTAL. It assumes that it is a mandatory hardware component.
However, this requirement is not specified in the **Limitations** section of the Core's README:
https://github.com/SiliconLabs/arduino#limitations

<br>
<br>

[image]

The request is:
---------------
Is it possible to avoid using of the RTC timer for the purpose of Arduino delay() ... and  millis() ... functions in the future releases of the Core ? Arduino Cores for other SoCs typically don't do that.

leonardocavagnis, 2024-09-05:
> @silabs-bozont
> 
> Thank you!
> 
> One more issue to talk about - when **32768 Hz XTAL is not connected** to PD0 and PD1 pins -
> 
> * delay() and delayMicroseconds()
> * millis() and micros()
>   are **operating approx. 100-110 times slower than they should**.
> 
> I found out that the Core relies heavily on the **external** 32.768 kHz XTAL. It assumes that it is a mandatory hardware component. However, this requirement is not specified in the **Limitations** section of the Core's README: https://github.com/SiliconLabs/arduino#limitations
> 
> 
> 
> <img alt="image" width="716" src="https://private-user-images.githubusercontent.com/5849637/360510344-7a269822-1082-48b8-be43-fc9be097d923.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU1NDM0MTcsIm5iZiI6MTcyNTU0MzExNywicGF0aCI6Ii81ODQ5NjM3LzM2MDUxMDM0NC03YTI2OTgyMi0xMDgyLTQ4YjgtYmU0My1mYzliZTA5N2Q5MjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MDVUMTMzMTU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTcxODllMDYwZGU2YTYxN2RkMTAxYzM1ZjlkNjMxODg1NTRkZmViMzI0ZmUyNDNkMjg5YzU2NzVmYWQ2MzdlYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.n-pE9XWfr0IZnO88lXie_Yh9pWw4ci5yXeakclR9adw">
> ## The request is:
> Is it possible to avoid using of the RTC timer for the purpose of Arduino delay() ... and millis() ... functions in the future releases of the Core ? Arduino Cores for other SoCs typically don't do that.

Hi @lyusupov,
Thanks for reporting.
Please open a new issue for better management.

lyusupov, 2024-09-05:
@leonardocavagnis 
> Please open a new issue for better management.

Done in https://github.com/SiliconLabs/arduino/issues/87

lyusupov, 2024-11-29:
https://github.com/SiliconLabs/arduino/releases/tag/2.2.0

closed 2024-11-29 by lyusupov