SiliconLabsSoftware/arduino #83: [request] watchdog (reset) function is missing in the API

lyusupov, 2024-08-27:
### Summary

Describe your feature request in a few lines### Summary

### Hardware

BGM220

### Core version

2.1.0

### Arduino IDE version

any

### Operating system

any

### Radio stack variant

No radio

### OpenThread Border Router device (if using Matter)

none

### Issue description

Current version of the API does not provide an interface that would allow to:
- init hardware watchdog timer
- reset hardware watchdog timer counter
- stop (de-init) hardware watchdog timer

[image]

leonardocavagnis, 2024-09-12:
Hello @lyusupov,
Thanks for the suggestion. Watchdog feature will be added in the next release.

lyusupov, 2024-11-29:
https://github.com/SiliconLabs/arduino/blob/main/libraries/WatchdogTimer/readme.md

https://github.com/SiliconLabs/arduino/releases/tag/2.2.0

closed 2024-11-29 by lyusupov