SiliconLabsSoftware/arduino #84: [bug] failure to perform I2C scan of device(s) connected over custom pins

lyusupov, 2024-08-30:
### Hardware

BGM220

### Core version

2.1.0

### Arduino IDE version

any

### Operating system

any

### Radio stack variant

No radio

### OpenThread Border Router device (if using Matter)

none

### Issue description

I use an I2C module with 2 devices connected to SDA=D1 (PC0) and SCL=D3 (PC2)

This is the scanner sketch:

```
#include <Arduino.h>

#if 1
#include <Wire.h>
arduino::TwoWire Wire_test(SL_I2C_PERIPHERAL,
                           SL_I2C_PERIPHERAL_NUM,
                           gpioPortC,
                           2, /* D3 */
                           gpioPortC,
                           0, /* D1 */
                           arduino_i2c_config);
#else
#include <FlexWire.h>
FlexWire Wire_test = FlexWire(D1, D3, false);
#endif

void setup() {
  Wire_test.begin(); // Initialize the I2C bus as a master
  Serial.begin(38400); // Initialize the serial communication at 115200 baud rate
  Serial.println("\nI2C Device Scanner"); // Print a header message
}

void loop() {
  byte error, address; // Variables to hold error status and I2C address
  int nDevices; // Variable to count the number of I2C devices found

  Serial.println("Scanning..."); // Print a message indicating the start of the scan
  nDevices = 0; // Reset device count

  // Loop through all possible I2C addresses (1 to 126)
  for(address = 1; address < 127; address++ ) {
    Wire_test.beginTransmission(address); // Start I2C transmission to the current address
    error = Wire_test.endTransmission(); // End the transmission and capture the error code

    if (error == 0) { // If no error occurred
      Serial.print("I2C device found at address 0x"); // Print the device found message
      if (address < 16) {
        Serial.print("0"); // Print a leading zero for addresses less than 0x10
      }
      Serial.println(address, HEX); // Print the address in hexadecimal format
      nDevices++; // Increment the device count
    }
    else if (error == 4) { // If an unknown error occurred
      Serial.print("Unknown error at address 0x"); // Print the error message
      if (address < 16) {
        Serial.print("0"); // Print a leading zero for addresses less than 0x10
      }
      Serial.println(address, HEX); // Print the address in hexadecimal format
    }
  }

  if (nDevices == 0) { // If no devices were found
    Serial.println("No I2C devices found\n"); // Print a message indicating no devices found
  }
  else { // If devices were found
    Serial.println("Scanning Completed\n"); // Print a message indicating the scan is done
  }

  delay(5000); // Wait 5 seconds before the next scan
}
```

When the sketch uses this Core **built-in Wire library** https://github.com/SiliconLabs/arduino/tree/main/libraries/Wire - it **FAIL**s:

```
I2C Device Scanner
Scanning...
I2C device found at address 0x01
I2C device found at address 0x02
I2C device found at address 0x03
I2C device found at address 0x04
I2C device found at address 0x05
I2C device found at address 0x06
I2C device found at address 0x07
I2C device found at address 0x08
I2C device found at address 0x09
I2C device found at address 0x0A
I2C device found at address 0x0B
I2C device found at address 0x0C
I2C device found at address 0x0D
I2C device found at address 0x0E
I2C device found at address 0x0F
I2C device found at address 0x10
I2C device found at address 0x11
I2C device found at address 0x12
I2C device found at address 0x13
I2C device found at address 0x14
I2C device found at address 0x15
I2C device found at address 0x16
I2C device found at address 0x17
I2C device found at address 0x18
I2C device found at address 0x19
I2C device found at address 0x1A
I2C device found at address 0x1B
I2C device found at address 0x1C
I2C device found at address 0x1D
I2C device found at address 0x1E
I2C device found at address 0x1F
I2C device found at address 0x20
I2C device found at address 0x21
I2C device found at address 0x22
I2C device found at address 0x23
I2C device found at address 0x24
I2C device found at address 0x25
I2C device found at address 0x26
I2C device found at address 0x27
I2C device found at address 0x28
I2C device found at address 0x29
I2C device found at address 0x2A
I2C device found at address 0x2B
I2C device found at address 0x2C
I2C device found at address 0x2D
I2C device found at address 0x2E
I2C device found at address 0x2F
I2C device found at address 0x30
I2C device found at address 0x31
I2C device found at address 0x32
I2C device found at address 0x33
I2C device found at address 0x34
I2C device found at address 0x35
I2C device found at address 0x36
I2C device found at address 0x37
I2C device found at address 0x38
I2C device found at address 0x39
I2C device found at address 0x3A
I2C device found at address 0x3B
I2C device found at address 0x3C
I2C device found at address 0x3D
I2C device found at address 0x3E
I2C device found at address 0x3F
I2C device found at address 0x40
I2C device found at address 0x41
I2C device found at address 0x42
I2C device found at address 0x43
I2C device found at address 0x44
I2C device found at address 0x45
I2C device found at address 0x46
I2C device found at address 0x47
I2C device found at address 0x48
I2C device found at address 0x49
I2C device found at address 0x4A
I2C device found at address 0x4B
I2C device found at address 0x4C
I2C device found at address 0x4D
I2C device found at address 0x4E
I2C device found at address 0x4F
I2C device found at address 0x50
I2C device found at address 0x51
I2C device found at address 0x52
I2C device found at address 0x53
I2C device found at address 0x54
I2C device found at address 0x55
I2C device found at address 0x56
I2C device found at address 0x57
I2C device found at address 0x58
I2C device found at address 0x59
I2C device found at address 0x5A
I2C device found at address 0x5B
I2C device found at address 0x5C
I2C device found at address 0x5D
I2C device found at address 0x5E
I2C device found at address 0x5F
I2C device found at address 0x60
I2C device found at address 0x61
I2C device found at address 0x62
I2C device found at address 0x63
I2C device found at address 0x64
I2C device found at address 0x65
I2C device found at address 0x66
I2C device found at address 0x67
I2C device found at address 0x68
I2C device found at address 0x69
I2C device found at address 0x6A
I2C device found at address 0x6B
I2C device found at address 0x6C
I2C device found at address 0x6D
I2C device found at address 0x6E
I2C device found at address 0x6F
I2C device found at address 0x70
I2C device found at address 0x71
I2C device found at address 0x72
I2C device found at address 0x73
I2C device found at address 0x74
I2C device found at address 0x75
I2C device found at address 0x76
I2C device found at address 0x77
I2C device found at address 0x78
I2C device found at address 0x79
I2C device found at address 0x7A
I2C device found at address 0x7B
I2C device found at address 0x7C
I2C device found at address 0x7D
I2C device found at address 0x7E
Scanning Completed
```

If I use the **FlexWire library** (software I2C) https://github.com/felias-fogg/FlexWire as a substitute for the built-in Wire library - **the sketch works nicely**

```
I2C Device Scanner
Scanning...
I2C device found at address 0x68
I2C device found at address 0x76
Scanning Completed
```

leonardocavagnis, 2024-09-05:
Hi @lyusupov,
Thanks for reporting.
The problem is in the `endTransmission()` function. It always returns success even though no values ​​are sent.

To workaround this issue, try to add a dummy write:
```
    Wire_test.beginTransmission(address); // Start I2C transmission to the current address
    Wire_test.write(0); //dummy write
    error = Wire_test.endTransmission(); // End the transmission and capture the error code
```

The output should be:
```
Scanning...
Unknown error at address 0x01
Unknown error at address 0x02
...
I2C device found at address 0x68
I2C device found at address 0x76
...
Unknown error at address 0x7D
Unknown error at address 0x7E
Scanning Completed
```

We'll address this issue in the next core release.

lyusupov, 2024-12-03:
I've rebuilt and re-flashed the I2C scan sketch above with recent Core 2.2.0

```
$ arduino-cli core list  | grep Sil
SiliconLabs:silabs  2.2.0     2.2.0  Silicon Labs
$ head ~/.arduino15/packages/SiliconLabs/hardware/silabs/2.2.0/libraries/Wire//library.properties
name=Wire
version=2.2.0
author=Silicon Labs
maintainer=Silicon Labs <arduino@silabs.com>
sentence=The standard Arduino Wire (I2C) library for Silicon Labs devices.
category=Communication
url=https://github.com/SiliconLabs/arduino
architectures=silabs
```

The issue still remains:

```
Scanning...
I2C device found at address 0x01
I2C device found at address 0x02
I2C device found at address 0x03
I2C device found at address 0x04
I2C device found at address 0x05
I2C device found at address 0x06
I2C device found at address 0x07
I2C device found at address 0x08
I2C device found at address 0x09
I2C device found at address 0x0A
I2C device found at address 0x0B
I2C device found at address 0x0C
I2C device found at address 0x0D
I2C device found at address 0x0E
I2C device found at address 0x0F
I2C device found at address 0x10
I2C device found at address 0x11
I2C device found at address 0x12
I2C device found at address 0x13
I2C device found at address 0x14
I2C device found at address 0x15
I2C device found at address 0x16
I2C device found at address 0x17
I2C device found at address 0x18
I2C device found at address 0x19
I2C device found at address 0x1A
I2C device found at address 0x1B
I2C device found at address 0x1C
I2C device found at address 0x1D
I2C device found at address 0x1E
I2C device found at address 0x1F
I2C device found at address 0x20
I2C device found at address 0x21
I2C device found at address 0x22
I2C device found at address 0x23
I2C device found at address 0x24
I2C device found at address 0x25
I2C device found at address 0x26
I2C device found at address 0x27
I2C device found at address 0x28
I2C device found at address 0x29
I2C device found at address 0x2A
I2C device found at address 0x2B
I2C device found at address 0x2C
I2C device found at address 0x2D
I2C device found at address 0x2E
I2C device found at address 0x2F
I2C device found at address 0x30
I2C device found at address 0x31
I2C device found at address 0x32
I2C device found at address 0x33
I2C device found at address 0x34
I2C device found at address 0x35
I2C device found at address 0x36
I2C device found at address 0x37
I2C device found at address 0x38
I2C device found at address 0x39
I2C device found at address 0x3A
I2C device found at address 0x3B
I2C device found at address 0x3C
I2C device found at address 0x3D
I2C device found at address 0x3E
I2C device found at address 0x3F
I2C device found at address 0x40
I2C device found at address 0x41
I2C device found at address 0x42
I2C device found at address 0x43
I2C device found at address 0x44
I2C device found at address 0x45
I2C device found at address 0x46
I2C device found at address 0x47
I2C device found at address 0x48
I2C device found at address 0x49
I2C device found at address 0x4A
I2C device found at address 0x4B
I2C device found at address 0x4C
I2C device found at address 0x4D
I2C device found at address 0x4E
I2C device found at address 0x4F
I2C device found at address 0x50
I2C device found at address 0x51
I2C device found at address 0x52
I2C device found at address 0x53
I2C device found at address 0x54
I2C device found at address 0x55
I2C device found at address 0x56
I2C device found at address 0x57
I2C device found at address 0x58
I2C device found at address 0x59
I2C device found at address 0x5A
I2C device found at address 0x5B
I2C device found at address 0x5C
I2C device found at address 0x5D
I2C device found at address 0x5E
I2C device found at address 0x5F
I2C device found at address 0x60
I2C device found at address 0x61
I2C device found at address 0x62
I2C device found at address 0x63
I2C device found at address 0x64
I2C device found at address 0x65
I2C device found at address 0x66
I2C device found at address 0x67
I2C device found at address 0x68
I2C device found at address 0x69
I2C device found at address 0x6A
I2C device found at address 0x6B
I2C device found at address 0x6C
I2C device found at address 0x6D
I2C device found at address 0x6E
I2C device found at address 0x6F
I2C device found at address 0x70
I2C device found at address 0x71
I2C device found at address 0x72
I2C device found at address 0x73
I2C device found at address 0x74
I2C device found at address 0x75
I2C device found at address 0x76
I2C device found at address 0x77
I2C device found at address 0x78
I2C device found at address 0x79
I2C device found at address 0x7A
I2C device found at address 0x7B
I2C device found at address 0x7C
I2C device found at address 0x7D
I2C device found at address 0x7E
Scanning Completed
```

dansiviter, 2024-12-11:
FYI I've raised this issue with Arduino Support.

leonardocavagnis, 2024-12-12:
Hey! It's not an issue, we decided to force "success" if no data is sent.

To check if an address is valid or not, add a dummy write:

```
    Wire_test.beginTransmission(address); // Start I2C transmission to the current address
    Wire_test.write(0); //dummy write
    error = Wire_test.endTransmission(); // End the transmission and capture the error code
```

leonardocavagnis, 2024-12-17:
@dansiviter Understood!
If the AHT20 sensor enters in an inconsistent state, what do you think to use an approach like [this](https://github.com/Seeed-Studio/Seeed_Arduino_AHT20/blob/master/src/ATH20.cpp#L13-L36)?

According to the [Seed_Arduino_AHT20](https://github.com/Seeed-Studio/Seeed_Arduino_AHT20) library, you can send a "start command" to the sensor and waiting for a reply within a time span.

silabs-bozont, 2025-05-07:
Hello @lyusupov,
The issue with you sketch is that it's not possible to dynamically remap the I2C pins by reinstantiating the Wire driver.
```
arduino::TwoWire Wire_test(SL_I2C_PERIPHERAL,
                           SL_I2C_PERIPHERAL_NUM,
                           gpioPortC,
                           2, /* D3 */
                           gpioPortC,
                           0, /* D1 */
                           arduino_i2c_config);
```
Creating an instance like this will not work - the constructor only tells the driver where the pins are mapped - but providing different pins than the mapping in the SDK config will cause it to not function.
Wire and Wire1 are instantiated with the correct SDK I2C peripheral mapping and therefore functional.
Does the scanning work if you use the default Wire or Wire1 instances?

lyusupov, 2025-05-08:
Hi @silabs-bozont 

[image]

The sketch runs on the E104 module (EFR32BG22)
https://www.cdebyte.com/products/E104-BT53A3#Specification

PD2 and PD3 GPIOs are not exposed there
https://www.cdebyte.com/products/E104-BT53A3/2#Pin