SiliconLabsSoftware/arduino #86: [bug] <title> Impossible to re-pair new device and bridge to homekit after first pair and delete

tiziano149, 2024-09-01:
### Hardware

Arduino Nano Matter

### Core version

2.1.0

### Arduino IDE version

2.3.2

### Operating system

macOS 14.5

### Radio stack variant

Matter

### OpenThread Border Router device (if using Matter)

Apple TV4K and HomePod Mini

### Issue description

After initially pairing the Arduino Nano Matter with HomeKit and properly decommissioning the device before deleting the bridge, it becomes impossible to re-pair the bridge to HomeKit. The first re-pairing attempt returns the expected “This accessory is not authorized” message, but subsequent attempts result in a timeout error, preventing the bridge from being added back to the HomeKit setup. This behavior is consistent across both Apple TV 4K and HomePod Mini.

Steps to Reproduce:

	1.	Pair the Arduino Nano Matter with HomeKit successfully.
	2.	Properly decommission the device from HomeKit.
	3.	Delete the bridge from HomeKit.
	4.	Attempt to re-pair the same Arduino Nano Matter bridge with HomeKit.
	5.	The first attempt results in “This accessory is not authorized” (as expected).
	6.	Subsequent attempts result in a timeout and the error “Accessory Cannot Be Added - Failed to establish a link with the accessory.”

Expected Behavior:
The bridge should re-pair with HomeKit without any issues after being decommissioned and deleted, even if the first attempt is unauthorized.

Actual Behavior:
The first re-pairing attempt shows “This accessory is not authorized,” but the subsequent attempts time out and fail with the error message “Accessory Cannot Be Added - Failed to establish a link with the accessory.”

Environment:

	•	Arduino Nano Matter
	•	HomeKit
	•	Apple TV 4K and HomePod Mini
	•	iOS version: 17.6.1

Additional Context:
This issue occurs on both Apple TV 4K and HomePod Mini, indicating that the problem is not specific to a particular device. It seems related to how HomeKit handles re-pairing after the initial decommissioning and deletion process.

Nephylie, 2024-09-03:
I have the same problem with my home assistant and a GL-S200 Thread border router.

leonardocavagnis, 2024-09-05:
Hi,
Did you successfully manage the decommissioning process on the Arduino Nano Matter side?
You can find more details [here](https://github.com/SiliconLabs/arduino/tree/main/libraries/Matter#unpairing--decommissioning).

tiziano149, 2024-09-05:
> Hi,
> 
> Did you successfully manage the decommissioning process on the Arduino Nano Matter side?
> 
> You can find more details [here](https://github.com/SiliconLabs/arduino/tree/main/libraries/Matter#unpairing--decommissioning).

Hi,
Thanks for the reply.
Yes, as stated in the first sentence.

leonardocavagnis, 2024-09-05:
Did you try using new provisioning data? 
https://github.com/SiliconLabs/arduino/tree/main/extra/arduino_matter_provision

tiziano149, 2024-09-05:
> Did you try using new provisioning data? 
> 
> https://github.com/SiliconLabs/arduino/tree/main/extra/arduino_matter_provision

Yup, tried all the different codes.
As it allows to connect to a new border router but then no more than once to the same one, it seems to me more of a caching issue on HomeKits side. Or some credential being stored and the causing the issue. But no idea how to get rid of it...

Nephylie, 2024-09-05:
I haven't tried it yet, but I found this :
https://support.apple.com/en-us/102135

leonardocavagnis, 2024-09-12:
Hi,
I replicated your setup and everything worked fine. I'm able to commission and decommission the same Nano Matter without any issues.
```
Arduino Nano Matter
HomePod Mini (A2374)
iPhone 12 (iOS 17.6.1)
HomeKit app
```

Please, could you clarify the step 2? To decommission it, I only perform step 3 (see attachments).
```
2.	Properly decommission the device from HomeKit.
3.	Delete the bridge from HomeKit.
```

[image]
[image]