SiliconLabsSoftware/arduino #88: [bug] device unable to connect after being reprogrammed

leonardpitzu, 2024-09-10:
### Hardware

SparkFun Thing Plus Matter - MGM240P

### Core version

2.1.0

### Arduino IDE version

2.3.2

### Operating system

macOS 14.6.1

### Radio stack variant

Matter

### OpenThread Border Router device (if using Matter)

HomePod mini

### Issue description

After flashing the board with the compiled sketch I am able to commission everything as expected. After refreshing the sketch the board appears as commissioned but does not connect to the boarder router any more (nothing changes in the setup otherwise). I have to manually remove the accessory from the iOS Home app, refresh the board and do the commissioning all over again. And this every time, for every code change!

leonardocavagnis, 2024-09-17:
Hi @leonardpitzu,
Thanks for your report. 
I tried to replicate your setup, and everything worked fine. First, I flashed one of the examples from the library, then I applied some modifications (e.g., adding Serial prints) and re-flashed the sketch.

Could you please provide us with some additional details about the changes you applied?

leonardpitzu, 2024-09-19:
It happens with any sample and reflashing the same, without any modifications, results in the behavior described. I think it has something to do with iOS 18 and its ability to connect to matter devices over thread without the need of an edge router. Could you please try and replicate this with iOS 18 and its default Hone app?

leonardpitzu, 2024-09-19:
Did some more testing today with the fully reset board and the temp sensor demo.
Steps to reproduce:
1. Flash the board with the demo
2. Pair the board with the iOS18 Home app
3. Test the sensor working
4. Release the board with the same demo (unchanged) or simply unplug the board
5. Thread connects but the demo hangs at “waiting for Matter discovery”