SiliconLabsSoftware/arduino #96: [request] I2C OLED display example

lboue, 2024-11-03:
### Summary

Hello,

I would use I2C OLED display example to print pairing QRcode. Could you add an example?

Regards

lboue, 2024-11-15:
I managed to run a demo code with an I2C OLED screen : `SSD1306_128X64`:

```
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>
#include "smiley_face.h" 

// QWIIC connector
#define SDA 20
#define SCL 19

// SW_I2C	Software emulated I2C/TWI
U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);

int xx = 20; 
int yy = 10; 

void setup() {
    Serial.begin(115200); // Initialize the serial communication at 115200 baud rate
    u8g2.begin();
}

void loop() {
    smeil_display();
    delay(500); 
}

void smeil_display() {
    const unsigned char* smileImages[] = {
        semil1, semil2, semil3, semil4, semil5,
        semil6, semil7, semil8, semil9, semil10,
        semil11, semil12, semil13, semil14, semil15,
        semil16, semil17, semil18, semil19, semil20,
        semil21, semil22, semil23, semil24, semil25,
        semil26, semil27
    };

    int delays[] = {
        40, 50, 40, 40, 40,
        40, 40, 50, 40, 40,
        40, 40, 40, 50, 40,
        40, 50, 40, 40, 50,
        40, 50, 40, 40, 50,
        50, 50, 40, 50
    };

    for (int i = 0; i < sizeof(smileImages) / sizeof(smileImages[0]); i++) {
        u8g2.firstPage();
        do {
            u8g2.drawXBMP(xx, yy, 48, 48, smileImages[i]); 
        } while (u8g2.nextPage());
        delay(delays[i]); 
    }
}
```

leonardocavagnis, 2024-11-25:
Hi @lboue,
For requests like this, we recommend using the Arduino Forum, which is better suited for community discussions and support. The Issues section is primarily intended for reporting bugs or requesting new features related to the core functionalities.
There is a dedicated section for the Arduino Nano Matter board [here](https://forum.arduino.cc/c/hardware/nano-family/nano-matter/195).
Feel free to post your question there, and the community will be happy to assist you.

commit 9bff82cd (2024-08-01): Release 2.1.0

* Add the released 2.0.0 version to the package index JSON

Also add the new and upcoming 2.1.0 version for local installs

* Bump the core version to 2.1.0

* Initialize the ADC input pin correctly

Without setting the ADC pin explicitly as an input we got
strange behavior on some of the pins.
Fixes #96

* remove debug print

* Add AES CCM support to the BLE variants

* Add NVM3 support to the 'no radio' variants

* Generate SDK files for the BLE and 'no radio' variants

* Implement NumberOfPrimaries support for Matter Color Control

* Implement direct copying to the 'variants' [... cut]
closed 2024-11-25 by lboue