SiliconLabsSoftware/arduino #97: [bug] I2C OLED display build error

lboue, 2024-11-03:
### Hardware

Silicon Labs xG24 Explorer Kit

### Core version

2.1.0

### Arduino IDE version

2.3.3

### Operating system

Windows 11

### Radio stack variant

Matter

### Issue description

Detailed description of the issue

Hello,

I would use I2C OLED display example to print pairing QRcode with my [Silicon Labs xG24 Explorer Kit](https://www.silabs.com/development-tools/wireless/efr32xg24-explorer-kit).
I am using the [Adafruit_SSD1306 library](https://github.com/adafruit/Adafruit_SSD1306/)

```
c:\Users\ludov\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp: In member function 'bool Adafruit_SSD1306::begin(uint8_t, uint8_t, bool, bool)':
c:\Users\ludov\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:530:44: error: 'digitalPinToPort' was not declared in this scope; did you mean 'digitalPinToInterrupt'?
  530 |     dcPort = (PortReg *)portOutputRegister(digitalPinToPort(dcPin));
      |                                            ^~~~~~~~~~~~~~~~
      |                                            digitalPinToInterrupt
c:\Users\ludov\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:530:25: error: 'portOutputRegister' was not declared in this scope
  530 |     dcPort = (PortReg *)portOutputRegister(digitalPinToPort(dcPin));
      |                         ^~~~~~~~~~~~~~~~~~
c:\Users\ludov\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:531:17: error: 'digitalPinToBitMask' was not declared in this scope
  531 |     dcPinMask = digitalPinToBitMask(dcPin);
      |                 ^~~~~~~~~~~~~~~~~~~

Using library SPI at version 2.1.0 in folder: C:\Users\ludov\AppData\Local\Arduino15\packages\SiliconLabs\hardware\silabs\2.1.0\libraries\SPI 
Using library Wire at version 2.1.0 in folder: C:\Users\ludov\AppData\Local\Arduino15\packages\SiliconLabs\hardware\silabs\2.1.0\libraries\Wire 
Using library Adafruit GFX Library at version 1.11.11 in folder: C:\Users\ludov\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Adafruit BusIO at version 1.16.2 in folder: C:\Users\ludov\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Adafruit SSD1306 at version 2.5.13 in folder: C:\Users\ludov\Documents\Arduino\libraries\Adafruit_SSD1306 
exit status 1
```

I'm trying to decide whether to correct it in the code or in the screen library.

Regards

### Minimal reproducer code

https://github.com/adafruit/Adafruit_SSD1306/blob/master/examples/ssd1306_128x64_i2c/ssd1306_128x64_i2c.ino

lboue, 2024-11-15:
I found another Library: 
https://github.com/olikraus/U8g2_Arduino

leonardocavagnis, 2024-11-15:
Hi @lboue,
It seems that the issue is related to the screen library.

Please modify this line
https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.h#L57
to:
`!defined(ARDUINO_ARCH_MBED) && !defined(ARDUINO_ARCH_RP2040) && !defined(ARDUINO_SILABS)`
and give it a try.

Let me know if it works! It should compile, but I'm not sure about the screen functionality (unfortunately, I don't have this screen to test 😢 ).

Thanks!

lboue, 2024-11-15:
Thanks, it works

[image]

leonardocavagnis, 2024-11-15:
Thank u too!
I opened a PR on Adafruit library to fix it https://github.com/adafruit/Adafruit_SSD1306/pull/279

lyusupov, 2024-12-02:
@leonardocavagnis 

> Please modify this line
> https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.h#L57
> to:
> `!defined(ARDUINO_ARCH_MBED) && !defined(ARDUINO_ARCH_RP2040) && !defined(ARDUINO_SILABS)`

I believe that you mean `ARDUINO_ARCH_SILABS` , right ?

`ARDUINO_SILABS` contains a string that introduces version numbers of this Core.
https://github.com/SiliconLabs/arduino/blob/main/platform.txt#L35

leonardocavagnis, 2024-12-03:
@lyusupov Good catch! thanks!
Please modify and let me know if it works.

lyusupov, 2024-12-03:
@leonardocavagnis 
would you mind to update [**your PR to Adafruit**](https://github.com/adafruit/Adafruit_SSD1306/pull/279) with `ARDUINO_ARCH_SILABS` symbol ?

leonardocavagnis, 2024-12-03:
@lyusupov sure!

silabs-bozont, 2024-12-03:
> @leonardocavagnis
>
>
> I believe that you mean `ARDUINO_ARCH_SILABS` , right ?
>
> `ARDUINO_SILABS` contains a string that introduces version numbers of this Core. [...]

I don't think it needs to be modified - it's correct as it is.
We use `ARDUINO_SILABS` as the platform discriminator - `ARDUINO_ARCH_SILABS` is not defined.
While the `ARDUINO_SILABS` macro has the value of the current core version `#ifdef` will definietly find it defined.
Here's my patch in Adafruit's BusIO library which added compatibility for example: https://github.com/adafruit/Adafruit_BusIO/pull/124/files

lyusupov, 2024-12-03:
@silabs-bozont 

> `ARDUINO_ARCH_SILABS` is not defined.

This does not look look like 100% accurate statement

[image]