SiliconLabsSoftware/arduino #98: [request] Support for Seeed Xiao MG24

rei-vilo, 2024-11-07:
### Summary

Seeed has announced the release of the [Xiao MG24](https://www.seeedstudio.com/Seeed-Studio-XIAO-MG24-p-6247.html) and the [Xiao MG24 Sense](https://www.seeedstudio.com/Seeed-XIAO-MG24-Sense-p-6248.html). 

Do you plan to support those two new boards?

Thank you!

lboue, 2024-11-14:
@rei-vilo The procedure can be found here, but is not applicable as the new version has not yet been published:
 - [Add the XIAO MG24 Board](https://wiki.seeedstudio.com/xiao_mg24_getting_started/#add-board)

rei-vilo, 2024-11-15:
Thank you! 

Is there an ETA for the new version?

loozhengyuan, 2024-11-16:
@rei-vilo From Seeed Studio's Discord, it appears that they have plans to publish before they start shipping the new boards (estimated end-Nov).

lboue, 2024-11-30:
@rei-vilo 
Check this: https://github.com/SiliconLabs/arduino/releases/tag/2.2.0

rei-vilo, 2024-12-02:
Thank you, case closed!

closed 2024-12-02 by rei-vilo