SiliconLabsSoftware/arduino #99: [bug] Build fails on `min()` and `max()`  when Matter library used

rei-vilo, 2024-11-14:
### Hardware

Arduino Nano Matter

### Core version

2.1.0

### Arduino IDE version

2.3.3

### Operating system

Debian

### Radio stack variant

Matter

### Issue description

Build fails on `min()` and `max()`  when the Matter library is used.

> Compilation error: call of overloaded 'min(uint16_t&, uint16_t&)' is ambiguous

See minimal code and cnsole output attached.

### Serial output

```
/home/user/.arduino15/packages/SiliconLabs/tools/gcc-arm-none-eabi/12.2.rel1/bin/arm-none-eabi-g++ -mcpu=cortex-m33 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=hard -c -g -w -std=c++11 -std=gnu++17 -Wall -Wextra -fno-exceptions -fdata-sections -ffunction-sections -fomit-frame-pointer -imacros sl_gcc_preinclude.h -mcmse -Wno-deprecated-declarations -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -fno-strict-aliasing -fno-unwind-tables -fno-asynchronous-unwind-tables -fno-common -Wno-sign-compare --specs=nano.specs -g -Os -DF_CPU=39000000 -DARDUINO=10607 "-DARDUINO_SILABS=\"2.1.0\"" -DARDUINO_NANO_MATTER -DARDUINO_ARCH_SILABS -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/cores/silabs/api/deprecated -DNUM_LEDS=3 -DNUM_BTN=1 -DNUM_HW_SERIAL=2 -DNUM_HW_SPI=2 -DNUM_HW_I2C=2 -DNUM_DAC_HW=2 -DARDUINO_MAIN_TASK_STACK_SIZE=2048 -DARDUINO_MATTER -DCHIP_CRYPTO_PLATFORM=1 -DIS_DEMO_LIGHT=1 -DNVM3_DEFAULT_MAX_OBJECT_SIZE=4092 -DNVM3_DEFAULT_NVM_SIZE=40960 -DSL_STATUS_LED=0 -D_WANT_REENT_SMALL=1 -DconfigNUM_THREAD_LOCAL_STORAGE_POINTERS=2 -DconfigNUM_USER_THREAD_LOCAL_STORAGE_POINTERS=0 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h> -DCHIP_HAVE_CONFIG_H=1 -DCURRENT_TIME_NOT_IMPLEMENTED=1 -DMBEDTLS_USER_CONFIG_FILE=<sli_psa_builtin_config.h> -DOPENTHREAD_CONFIG_DETERMINISTIC_ECDSA_ENABLE=0 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=0 -DSILABS_OTA_ENABLED=1 -DRTT_USE_ASM=0 -DENABLE_WSTK_LEDS=1 -DMGM240SD22VNA=1 -DSL_APP_PROPERTIES=1 -DHARDWARE_BOARD_DEFAULT_RF_BAND_2400=1 -DHARDWARE_BOARD_SUPPORTS_1_RF_BAND=1 -DHARDWARE_BOARD_SUPPORTS_RF_BAND_2400=1 "-DSL_BOARD_NAME=\"BRD2707A\"" "-DSL_BOARD_REV=\"A00\"" -DconfigNUM_SDK_THREAD_LOCAL_STORAGE_POINTERS=2 -DSL_COMPONENT_CATALOG_PRESENT=1 -DMBEDTLS_CONFIG_FILE=<sl_mbedtls_config.h> -DRADIO_CONFIG_DMP_SUPPORT=1 "-DOPENTHREAD_CORE_CONFIG_PLATFORM_CHECK_FILE=\"openthread-core-efr32-config-check.h\"" "-DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-core-efr32-config.h\"" "-DOPENTHREAD_CONFIG_FILE=\"sl_openthread_generic_config.h\"" -DOPENTHREAD_FTD=1 "-DSL_OPENTHREAD_STACK_FEATURES_CONFIG_FILE=\"sl_openthread_features_config.h\"" -DMBEDTLS_PSA_CRYPTO_CONFIG_FILE=<psa_crypto_config.h> -DSL_RAIL_LIB_MULTIPROTOCOL_SUPPORT=1 -DSL_RAIL_UTIL_PA_CONFIG_HEADER=<sl_rail_util_pa_config.h> -DSLI_RADIOAES_REQUIRES_MASKING=1 -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/autogen/zap-generated -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/autogen/zap-generated/app -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/config -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/config/btconf -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/config/common -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/autogen -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/include -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/examples/platform/silabs -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/include -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/lib -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/third_party/nlassert/repo/include -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/third_party/nlio/repo/include -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/zzz_generated/app-common -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/slc/inc -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/examples/platform/silabs/efr32 -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/platform/silabs/efr32 -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/examples/providers -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/basic-information -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/color-control-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/diagnostic-logs-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/door-lock-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/general-commissioning-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/general-diagnostics-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/groups-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/identify-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/level-control -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/network-commissioning -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/occupancy-sensor-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/on-off-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/ota-requestor -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/third_party/silabs/gecko_sdk/util/third_party/segger/systemview/SEGGER -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/software-diagnostics-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/switch-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/matter_2.2.0/src/app/clusters/window-covering-server -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/gecko_sdk_4.4.0/platform/Device/SiliconLabs/MGM24/Include -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/gecko_sdk_4.4.0/app/common/util/app_assert -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/gecko_sdk_4.4.0/app/common/util/app_log -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/gecko_sdk_4.4.0/platform/common/inc -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/gecko_sdk_4.4.0/protocol/bluetooth/inc -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/matter/gecko_sdk_4.4.0/protocol/bluetooth/bgstack/ll/inc -I/home/user/.arduino15/packages/SiliconLabs/hardware/silabs/2.1.0/variants/nano_matter/mat [... cut ...] ]'
[... 1 lines cut ...]
      |                                 ^~~
Compilation error: call of overloaded 'min(uint16_t&, uint16_t&)' is ambiguous
```

### Minimal reproducer code

#include <Matter.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  uint16_t a = 2;
  uint16_t b = 3;
  uint16_t c = min(a, b);
  uint16_t d = max(a, b);

  Serial.println(c);
  Serial.println(d);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
}

leonardocavagnis, 2024-11-15:
Hi @rei-vilo,
Thanks for the bug report!
Unfortunately, `min` and `max` functions are declared in both the [ArduinoCore-API](https://github.com/arduino/ArduinoCore-API/blob/master/api/Common.h#L143-L170) and [Matter](https://github.com/SiliconLabs/arduino/blob/main/variants/nano_matter/matter/matter_2.2.0/src/lib/support/CodeUtils.h#L133-L155) libraries.

If you want to use the functions alongside the `Matter.h` library, you can use namespace `chip::` like this:

```
#include <Matter.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  uint16_t a = 2;
  uint16_t b = 3;
  uint16_t c = chip::min(a, b);
  uint16_t d = chip::max(a, b);

  Serial.println(c);
  Serial.println(d);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
}
```

rei-vilo, 2024-11-15:
Thank you for the confirmation of the conflict on `min()`  and max()`. 

However, using `chip::min()` or `chip::max()` limits the code portability.

silabs-bozont, 2025-04-08:
Hey @rei-vilo,
Thank you for the report!
It looks like the Matter library pulls in the whole `chip` namespace - which also provides definitions for the `min` and `max` functions. This is not ideal - as you said your code becomes less portable.
As a solution I've limited the `chip` namespace usage to just include the necessary things for the Matter library and nothing else. The default will be the standard min/max function.
The fix will be included in the next release!

closed 2025-07-28 by silabs-bozont