SiliconLabsSoftware/arduino #104: [bug] I²C Scanning fails

dansiviter, 2024-12-11:
### Hardware

Arduino Nano Matter

### Core version

2.0.0, 2.1.0, 2.2.0

### Arduino IDE version

2.3.4

### Operating system

Windows 11

### Radio stack variant

Matter

### Issue description

Using typical I²C scanning (e.g. [arduino.cc/I2cScanner](https://playground.arduino.cc/Main/I2cScanner/)) does not work. It seems the implementation of [#endTransmission(...)](https://github.com/SiliconLabs/arduino/blob/2.2.0/libraries/Wire/Wire.cpp#L226) assumes `SUCCESS` in most circumstances which appears incorrect.
This has been tried on multiple boards and using both `Wire` and `Wire1` so assuming the code is at fault.

### Serial output

```
Scanning...
I2C device found at address 0x01  !
I2C device found at address 0x02  !
I2C device found at address 0x03  !
I2C device found at address 0x04  !
I2C device found at address 0x05  !
I2C device found at address 0x06  !
... truncated for brevity ...
I2C device found at address 0x79  !
I2C device found at address 0x7A  !
I2C device found at address 0x7B  !
I2C device found at address 0x7C  !
I2C device found at address 0x7D  !
I2C device found at address 0x7E  !
done
```

### Minimal reproducer code

Use [arduino.cc/I2cScanner](https://playground.arduino.cc/Main/I2cScanner/)) example.

lyusupov, 2024-12-11:
Duplicates https://github.com/SiliconLabs/arduino/issues/84

dansiviter, 2024-12-11:
Thanks @lyusupov. I'll close in preference to #84.

closed 2024-12-11 by dansiviter