SiliconLabsSoftware/arduino #110: [request] Change in ArduinoLowPower DeepSleepExternalWakeUp example

SebKister, 2025-01-04:
### Summary

The current example demonstrating the external wake up from deep sleep uses the pin PC0. Unfortunately, this is also defined as the DEEP_SLEEP_ESCAPE_PIN in the escape_hatch function. If the user holds the wake-up switch low too long, it will not provoke a wake up but trigger the escape mechanism. The example could use the PA5 pin for example and clarify the role of PC0 in the wake up exit mechanism.

SebKister, 2025-01-07:
@lboue I use a Seeed Studio XIAO MG24 Sense

silabs-bozont, 2025-04-08:
Hello @SebKister,
Thank you for the bug report! It looks like you're correct - on the XIAO MG24 the deep sleep wake up pin and the escape pin are the same - PC0 (D0). 
PC0 is used in the wake-up examples because it's a dedicated wake-up pin and is exposed on all the boards - therefore the example should be left as is.
For the escape pin any pin can be used, it doesn't need to be a dedicated wake up pin.
I've reconfigured the XIAO MG24 to use PC1 (D1) as its escape pin - it doesn't seem to conflict with anything - and now waking up the device doesn't also trigger the escape mechanism.
We'll include this change in the next release. Let me know what you think!